AdnanHodzic/displaylink-debian #710: Blank Screen, Can't -setprovideroutputsource

arturogatti, 2022-04-24:
xrandr --setprovideroutputsource 2 0
X Error of failed request: BadValue (integer parameter out of range for operation)
Major opcode of failed request: 140 (RANDR)
Minor opcode of failed request: 35 (RRSetProviderOutputSource)
Value in failed request: 0x43
Serial number of failed request: 19
Current serial number in output stream: 20

X Error of failed request: BadValue (integer parameter out of range for operation)
Major opcode of failed request: 140 (RANDR)
Minor opcode of failed request: 35 (RRSetProviderOutputSource)
Value in failed request: 0x43
Serial number of failed request: 19
Current serial number in output stream: 20

-------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-10-amd64

---------------- DisplayLink info ----------------

Driver version: 1.10.0
1.10.1
1.9.0+dfsg
DisplayLink service status: up and running
EVDI service version: 1.10.1

------------------ Graphics card -----------------

Vendor: nvidia
Subsystem: [NVS
GT218
VGA: NVIDIA Corporation GF119 [NVS 310] (rev a1)
NVIDIA Corporation GT218 [GeForce 210] (rev a2)
VGA (3D):
X11 version: 1.20.11-1+deb11u1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf.d/20-intel.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
Section "Device"
Identifier "DisplayLink"
Driver "modesetting"
Option "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x43 cap: 0x0 crtcs: 2 outputs: 2 associated providers: 0 name:modesetting
Provider 1: id: 0xe0 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0xbe cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x9c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x6f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-----

sudo dmesg | grep Display
[ 2.356669] usb 2-10: Manufacturer: DisplayLink
[ 55.152164] evdi: [I] (card0) Opened by Task 1456 (DesktopManagerE) of process 1356 (DisplayLinkMana)
[ 55.432615] evdi: [I] (card0) Connected with Task 1456 (DesktopManagerE) of process 1356 (DisplayLinkMana)

----

systemctl status dlm.service failure
Unit dlm.service could not be found.
Unit failure.service could not be found.

arturogatti, 2022-04-24:
here is the output of: xrandr -q --verbose

Screen 0: minimum 320 x 200, current 3360 x 1050, maximum 65535 x 65535
DP-1 connected 1680x1050+0+0 (0x44) normal (normal left inverted right x axis y axis) 473mm x 296mm
	Identifier: 0x41
	Timestamp:  25878472
	Subpixel:   unknown
	Gamma:      1.0:1.0:1.0
	Brightness: 1.0
	Clones:    
	CRTC:       1
	CRTCs:      1 0
	Transform:  1.000000 0.000000 0.000000
	            0.000000 1.000000 0.000000
	            0.000000 0.000000 1.000000
	           filter: 
	_MUTTER_PRESENTATION_OUTPUT: 0 
	EDID: 
		00ffffffffffff0010ac42f04d593844
		0b170103802f1e78ea23c5a5574f9e26
		0f5054a54b00714f8180b30001010101
		01010101010121399030621a274068b0
		3600d9281100001c000000ff00465030
		34463333454438594d0a000000fc0044
		454c4c2050323231330a2020000000fd
		00384b1e5310000a2020202020200090
	link-status: Good 
		supported: Good, Bad
	CONNECTOR_ID: 40 
		supported: 40
	non-desktop: 0 
		range: (0, 1)
  1680x1050 (0x44) 146.250MHz -HSync +VSync *current
        h: width  1680 start 1784 end 1960 total 2240 skew    0 clock  65.29KHz
        v: height 1050 start 1053 end 1059 total 1089           clock  59.95Hz
  1280x1024 (0x45) 135.000MHz +HSync +VSync
        h: width  1280 start 1296 end 1440 total 1688 skew    0 clock  79.98KHz
        v: height 1024 start 1025 end 1028 total 1066           clock  75.02Hz
  1280x1024 (0x46) 108.000MHz +HSync +VSync
        h: width  1280 start 1328 end 1440 total 1688 skew    0 clock  63.98KHz
        v: height 1024 start 1025 end 1028 total 1066           clock  60.02Hz
  1152x864 (0x47) 108.000MHz +HSync +VSync
        h: width  1152 start 1216 end 1344 total 1600 skew    0 clock  67.50KHz
        v: height  864 start  865 end  868 total  900           clock  75.00Hz
  1024x768 (0x48) 78.750MHz +HSync +VSync
        h: width  1024 start 1040 end 1136 total 1312 skew    0 clock  60.02KHz
        v: height  768 start  769 end  772 total  800           clock  75.03Hz
  1024x768 (0x49) 65.000MHz -HSync -VSync
        h: width  1024 start 1048 end 1184 total 1344 skew    0 clock  48.36KHz
        v: height  768 start  771 end  777 total  806           clock  60.00Hz
  800x600 (0x4a) 49.500MHz +HSync +VSync
        h: width   800 start  816 end  896 total 1056 skew    0 clock  46.88KHz
        v: height  600 start  601 end  604 total  625           clock  75.00Hz
  800x600 (0x4b) 40.000MHz +HSync +VSync
        h: width   800 start  840 end  968 total 1056 skew    0 clock  37.88KHz
        v: height  600 start  601 end  605 total  628           clock  60.32Hz
  640x480 (0x4c) 31.500MHz -HSync -VSync
        h: width   640 start  656 end  720 total  840 skew    0 clock  37.50KHz
        v: height  480 start  481 end  484 total  500           clock  75.00Hz
  640x480 (0x4d) 25.175MHz -HSync -VSync
        h: width   640 start  656 end  752 total  800 skew    0 clock  31.47KHz
        v: height  480 start  490 end  492 total  525           clock  59.94Hz
DP-2 connected primary 1680x1050+1680+0 (0x44) normal (normal left inverted right x axis y axis) 473mm x 296mm
	Identifier: 0x42
	Timestamp:  25878472
	Subpixel:   unknown
	Gamma:      1.0:1.0:1.0
	Brightness: 1.0
	Clones:    
	CRTC:       0
	CRTCs:      1 0
	Transform:  1.000000 0.000000 0.000000
	            0.000000 1.000000 0.000000
	            0.000000 0.000000 1.000000
	           filter: 
	_MUTTER_PRESENTATION_OUTPUT: 0 
	EDID: 
		00ffffffffffff0010ac42f04d5a5541
		0c170103802f1e78ea23c5a5574f9e26
		0f5054a54b00714f8180b30001010101
		01010101010121399030621a274068b0
		3600d9281100001c000000ff00464a34
		344a33334c41555a4d0a000000fc0044
		454c4c2050323231330a2020000000fd
		00384b1e5310000a2020202020200050
	link-status: Good 
		supported: Good, Bad
	CONNECTOR_ID: 42 
		supported: 42
	non-desktop: 0 
		range: (0, 1)
  1680x1050 (0x44) 146.250MHz -HSync +VSync *current
        h: width  1680 start 1784 end 1960 total 2240 skew    0 clock  65.29KHz
        v: height 1050 start 1053 end 1059 total 1089           clock  59.95Hz
  1280x1024 (0x45) 135.000MHz +HSync +VSync
        h: width  1280 start 1296 end 1440 total 1688 skew    0 clock  79.98KHz
        v: height 1024 start 1025 end 1028 total 1066           clock  75.02Hz
  1280x1024 (0x46) 108.000MHz +HSync +VSync
        h: width  1280 start 1328 end 1440 total 1688 skew    0 clock  63.98KHz
        v: height 1024 start 1025 end 1028 total 1066           clock  60.02Hz
  1152x864 (0x47) 108.000MHz +HSync +VSync
        h: width  1152 start 1216 end 1344 total 1600 skew    0 clock  67.50KHz
        v: height  864 start  865 end  868 total  900           clock  75.00Hz
  1024x768 (0x48) 78.750MHz +HSync +VSync
        h: width  1024 start 1040 end 1136 total 1312 skew    0 clock  60.02KHz
        v: height  768 start  769 end  772 total  800           clock  75.03Hz
  1024x768 (0x49) 65.000MHz -HSync -VSync
        h: width  1024 start 1048 end 1184 total 1344 skew    0 clock  48.36KHz
        v: height  768 start  771 end  777 total  806           clock  60.00Hz
  800x600 (0x4a) 49.500MHz +HSync +VSync
        h: width   800 start  816 end  896 total 1056 skew    0 clock  46.88KHz
        v: height  600 start  601 end  604 total  625           clock  75.00Hz
  800x600 (0x4b) 40.000MHz +HSync +VSync
        h: width   800 start  840 end  968 total 1056 skew    0 clock  37.88KHz
        v: height  600 start  601 end  605 total  628           clock  60.32Hz
  640x480 (0x4c) 31.500MHz -HSync -VSync
        h: width   640 start  656 end  720 total  840 skew    0 clock  37.50KHz
        v: height  480 start  481 end  484 total  500           clock  75.00Hz
  640x480 (0x4d) 25.175MHz -HSync -VSync
        h: width   640 start  656 end  752 total  800 skew    0 clock  31.47KHz
        v: height  480 start  490 end  492 total  525           clock  59.94Hz

Barabazs, 2022-04-25:
Can you try to remove `/etc/X11/xorg.conf.d/20-intel.conf`?
Just append .back to the filename

arturogatti, 2022-04-25:
still nothing:

--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-10-amd64

---------------- DisplayLink info ----------------

Driver version: 1.10.0
1.10.1
1.9.0+dfsg
DisplayLink service status: up and running
EVDI service version: 1.10.1

------------------ Graphics card -----------------

Vendor: nvidia
Subsystem: [NVS
GT218
VGA: NVIDIA Corporation GF119 [NVS 310] (rev a1)
NVIDIA Corporation GT218 [GeForce 210] (rev a2)
VGA (3D): 
X11 version: 1.20.11-1+deb11u1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
Identifier "DisplayLink"
Driver "modesetting"
Option "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x43 cap: 0x0 crtcs: 2 outputs: 2 associated providers: 0 name:modesetting
Provider 1: id: 0xe0 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0xbe cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x9c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x6f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

arturogatti, 2022-04-25:
xrandr --setprovideroutputsource 1 0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  140 (RANDR)
  Minor opcode of failed request:  35 (RRSetProviderOutputSource)
  Value in failed request:  0x43
  Serial number of failed request:  19
  Current serial number in output stream:  20

-------------------

sucks.

arturogatti, 2022-04-25:
uninstalled, rebooted, reinstalled, rebooted and now that screen is back to green.

Barabazs, 2022-04-26:
Have you tried with installing either the nvidia-lts or nivida-beta drivers?
That seems to be a frequent solution.

arturogatti, 2022-04-27:
the nvidia-lts rebooted machine into a blank screen, purged all nvidia + installed nvidia x86_64-390.147 driver (390.xx legacy driver was installed before). 

did a uninstall, reboot, reinstall of displaylink-debian and still getting a blank screen on that monitor - but now i am able to do xrandr --setprovideroutputsource

**current debug output:** 

--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-10-amd64

---------------- DisplayLink info ----------------

Driver version: 1.10.0
1.10.1
DisplayLink service status: up and running
EVDI service version: 1.10.1

------------------ Graphics card -----------------

Vendor: nvidia
Subsystem: [NVS
GT218
VGA: NVIDIA Corporation GF119 [NVS 310] (rev a1)
NVIDIA Corporation GT218 [GeForce 210] (rev a2)
VGA (3D): 
X11 version: 1.20.11-1+deb11u1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x279 cap: 0x1, Source Output crtcs: 2 outputs: 4 associated providers: 4 name:NVIDIA-0
Provider 1: id: 0x31c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x2fa cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x2d8 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x2ab cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

------------------------

...let me know

arturogatti, 2022-04-27:
let me know if i set the providers wrong, etc.

arturogatti, 2022-04-27:
**output of sudo dmesg | grep Display**

[    2.153269] usb 2-10: Manufacturer: DisplayLink
[   52.556578] evdi: [I] (card0) Opened by Task 1470 (DesktopManagerE) of process 1372 (DisplayLinkMana)
[   52.582952] evdi: [I] (card0) Connected with Task 1470 (DesktopManagerE) of process 1372 (DisplayLinkMana)

arturogatti, 2022-04-28:
look @  output of **xrandr --verbose**

-------------------

Screen 0: minimum 8 x 8, current 3360 x 1050, maximum 16384 x 16384
DP-0 connected 1680x1050+0+0 (0x27d) normal (normal left inverted right x axis y axis) 473mm x 296mm
	Identifier: 0x27c
	Timestamp:  109692
	Subpixel:   unknown
	Gamma:      1.0:1.0:1.0
	Brightness: 1.0
	Clones:    
	CRTC:       1
	CRTCs:      1 0
	Transform:  1.000000 0.000000 0.000000
	            0.000000 1.000000 0.000000
	            0.000000 0.000000 1.000000
	           filter: 
	_MUTTER_PRESENTATION_OUTPUT: 0 
	CscMatrix: 65536 0 0 0 0 65536 0 0 0 0 65536 0 
	EDID: 
		00ffffffffffff0010ac42f04d593844
		0b170103802f1e78ea23c5a5574f9e26
		0f5054a54b00714f8180b30001010101
		01010101010121399030621a274068b0
		3600d9281100001c000000ff00465030
		34463333454438594d0a000000fc0044
		454c4c2050323231330a2020000000fd
		00384b1e5310000a2020202020200090
	BorderDimensions: 4 
		supported: 4
	Border: 0 0 0 0 
		range: (0, 65535)
	SignalFormat: TMDS 
		supported: TMDS
	ConnectorType: DisplayPort 
	ConnectorNumber: 0 
	_ConnectorLocation: 1 
	non-desktop: 0 
		supported: 0, 1
  1680x1050 (0x27d) 146.250MHz -HSync +VSync *current +preferred
        h: width  1680 start 1784 end 1960 total 2240 skew    0 clock  65.29KHz
        v: height 1050 start 1053 end 1059 total 1089           clock  59.95Hz
  1280x1024 (0x27e) 135.000MHz +HSync +VSync
        h: width  1280 start 1296 end 1440 total 1688 skew    0 clock  79.98KHz
        v: height 1024 start 1025 end 1028 total 1066           clock  75.02Hz
  1280x1024 (0x27f) 108.000MHz +HSync +VSync
        h: width  1280 start 1328 end 1440 total 1688 skew    0 clock  63.98KHz
        v: height 1024 start 1025 end 1028 total 1066           clock  60.02Hz
  1152x864 (0x280) 108.000MHz +HSync +VSync
        h: width  1152 start 1216 end 1344 total 1600 skew    0 clock  67.50KHz
        v: height  864 start  865 end  868 total  900           clock  75.00Hz
  1024x768 (0x281) 78.750MHz +HSync +VSync
        h: width  1024 start 1040 end 1136 total 1312 skew    0 clock  60.02KHz
        v: height  768 start  769 end  772 total  800           clock  75.03Hz
  1024x768 (0x282) 65.000MHz -HSync -VSync
        h: width  1024 start 1048 end 1184 total 1344 skew    0 clock  48.36KHz
        v: height  768 start  771 end  777 total  806           clock  60.00Hz
  800x600 (0x283) 49.500MHz +HSync +VSync
        h: width   800 start  816 end  896 total 1056 skew    0 clock  46.88KHz
        v: height  600 start  601 end  604 total  625           clock  75.00Hz
  800x600 (0x284) 40.000MHz +HSync +VSync
        h: width   800 start  840 end  968 total 1056 skew    0 clock  37.88KHz
        v: height  600 start  601 end  605 total  628           clock  60.32Hz
  640x480 (0x285) 31.500MHz -HSync -VSync
        h: width   640 start  656 end  720 total  840 skew    0 clock  37.50KHz
        v: height  480 start  481 end  484 total  500           clock  75.00Hz
  640x480 (0x286) 25.175MHz -HSync -VSync
        h: width   640 start  656 end  752 total  800 skew    0 clock  31.47KHz
        v: height  480 start  490 end  492 total  525           clock  59.94Hz
[... 254 lines cut ...]
        h: width  1280 start 1296 end 1440 total 1688 skew    0 clock  79.98KHz
        v: height 1024 start 1025 end 1028 total 1066           clock  75.02Hz
  1280x1024 (0x27f) 108.000MHz +HSync +VSync
        h: width  1280 start 1328 end 1440 total 1688 skew    0 clock  63.98KHz
        v: height 1024 start 1025 end 1028 total 1066           clock  60.02Hz
  1024x768 (0x281) 78.750MHz +HSync +VSync
        h: width  1024 start 1040 end 1136 total 1312 skew    0 clock  60.02KHz
        v: height  768 start  769 end  772 total  800           clock  75.03Hz
  1024x768 (0x282) 65.000MHz -HSync -VSync
        h: width  1024 start 1048 end 1184 total 1344 skew    0 clock  48.36KHz
        v: height  768 start  771 end  777 total  806           clock  60.00Hz
  800x600 (0x283) 49.500MHz +HSync +VSync
        h: width   800 start  816 end  896 total 1056 skew    0 clock  46.88KHz
        v: height  600 start  601 end  604 total  625           clock  75.00Hz
  800x600 (0x284) 40.000MHz +HSync +VSync
        h: width   800 start  840 end  968 total 1056 skew    0 clock  37.88KHz
        v: height  600 start  601 end  605 total  628           clock  60.32Hz
  640x480 (0x285) 31.500MHz -HSync -VSync
        h: width   640 start  656 end  720 total  840 skew    0 clock  37.50KHz
        v: height  480 start  481 end  484 total  500           clock  75.00Hz
  640x480 (0x286) 25.175MHz -HSync -VSync
        h: width   640 start  656 end  752 total  800 skew    0 clock  31.47KHz
        v: height  480 start  490 end  492 total  525           clock  59.94Hz

-----

arturogatti, 2022-04-28:
output of **xrandr --current** 

Screen 0: minimum 8 x 8, current 3360 x 1050, maximum 16384 x 16384
DP-0 connected 1680x1050+0+0 (normal left inverted right x axis y axis) 473mm x 296mm
   1680x1050     59.95*+
   1280x1024     75.02    60.02  
   1152x864      75.00  
   1024x768      75.03    60.00  
   800x600       75.00    60.32  
   640x480       75.00    59.94  
DP-1 connected primary 1680x1050+1680+0 (normal left inverted right x axis y axis) 473mm x 296mm
   1680x1050     59.95*+
   1280x1024     75.02    60.02  
   1152x864      75.00  
   1024x768      75.03    60.00  
   800x600       75.00    60.32  
   640x480       75.00    59.94  
DP-2 disconnected (normal left inverted right x axis y axis)
DP-3 disconnected (normal left inverted right x axis y axis)
DVI-I-4-4 disconnected (normal left inverted right x axis y axis)
DVI-I-3-3 disconnected (normal left inverted right x axis y axis)
DVI-I-2-2 disconnected (normal left inverted right x axis y axis)
DVI-I-1-1 connected (normal left inverted right x axis y axis)
   1920x1080     60.00 +
   1680x1050     59.95  
   1280x1024     75.02    60.02  
   1440x900      59.89  
   1280x960      60.00  
   1280x720      60.00  
   1024x768      75.03    70.07    60.00  
   832x624       74.55  
   800x600       72.19    75.00    60.32    56.25  
   640x480       75.00    72.81    66.67    59.94  
   720x400       70.08  
  1680x1050 (0x27d) 146.250MHz -HSync +VSync
        h: width  1680 start 1784 end 1960 total 2240 skew    0 clock  65.29KHz
        v: height 1050 start 1053 end 1059 total 1089           clock  59.95Hz
  1280x1024 (0x27e) 135.000MHz +HSync +VSync
        h: width  1280 start 1296 end 1440 total 1688 skew    0 clock  79.98KHz
        v: height 1024 start 1025 end 1028 total 1066           clock  75.02Hz
  1280x1024 (0x27f) 108.000MHz +HSync +VSync
        h: width  1280 start 1328 end 1440 total 1688 skew    0 clock  63.98KHz
        v: height 1024 start 1025 end 1028 total 1066           clock  60.02Hz
  1024x768 (0x281) 78.750MHz +HSync +VSync
        h: width  1024 start 1040 end 1136 total 1312 skew    0 clock  60.02KHz
        v: height  768 start  769 end  772 total  800           clock  75.03Hz
  1024x768 (0x282) 65.000MHz -HSync -VSync
        h: width  1024 start 1048 end 1184 total 1344 skew    0 clock  48.36KHz
        v: height  768 start  771 end  777 total  806           clock  60.00Hz
  800x600 (0x283) 49.500MHz +HSync +VSync
        h: width   800 start  816 end  896 total 1056 skew    0 clock  46.88KHz
        v: height  600 start  601 end  604 total  625           clock  75.00Hz
  800x600 (0x284) 40.000MHz +HSync +VSync
        h: width   800 start  840 end  968 total 1056 skew    0 clock  37.88KHz
        v: height  600 start  601 end  605 total  628           clock  60.32Hz
  640x480 (0x285) 31.500MHz -HSync -VSync
        h: width   640 start  656 end  720 total  840 skew    0 clock  37.50KHz
        v: height  480 start  481 end  484 total  500           clock  75.00Hz
  640x480 (0x286) 25.175MHz -HSync -VSync
        h: width   640 start  656 end  752 total  800 skew    0 clock  31.47KHz
        v: height  480 start  490 end  492 total  525           clock  59.94Hz

arturogatti, 2022-05-02:
....?

Barabazs, 2022-05-29:
It might be related to driver support issues for specific kernels. (anecdotally...) Your best bet is to open an issue on the nvidia forums or displaylink, as this is not related to `displaylink-debian`

closed 2022-05-29 by Barabazs