AdnanHodzic/displaylink-debian #712: Failed to start Light Display Manager. - Debian XFCE

alexanderepping, 2022-04-29:
I am running Debian 11 with XFCE and after running the installation and rebooting, I get the following message:
Failed to start Light Display Manager.

When trying to run xrandr, I just get "Can't open display" (which is also why I removed the lines in the debug function).
When removing /etc/X11/xorg.conf.d/20-displaylink.conf the system boots as usual but DisplayLink doesn't work.

My debug output (without the xrandr-stuff) is:
-------------------------------------------------------------------

Starting Debug ...

--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-13-amd64

---------------- DisplayLink info ----------------

Driver version: 1.10.1
DisplayLink service status: up and running
EVDI service version: 1.10.1

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation HD Graphics 620 (rev 02)
VGA (3D): NVIDIA Corporation GM108M [GeForce 940MX] (rev a2)
X11 version: 1.20.11-1+deb11u1
X11 configs: /etc/X11/xorg.conf.d/00-keyboard.conf /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

Barabazs, 2022-04-30:
Are you running this under a root account?

alexanderepping, 2022-04-30:
I am honestly not 100% certain, however I ran everything as my usual (and only) user (which therefore should have root privileges) using `sudo`. I also was never asked to set a root password and therefore didn't set one. 

So, I would say that yes, I am running this under a root account!

alexanderepping, 2022-04-30:
I think, I actually got it working now! 
My problem was actually the same as the one solved [here](https://github.com/AdnanHodzic/displaylink-debian/issues/228#issuecomment-467889348), which I wasn't expecting, since xrandr didn't even work for me and it seemed to me, like Lightdm was the problem. After changing the `/etc/X11/xorg.conf.d/20-displaylink.conf` file, my system booted. 
I will test it out when I am at home again. 
I hope I didn't waste your time and many thanks!

closed 2022-05-07 by alexanderepping