AdnanHodzic/displaylink-debian #713: Display showing wrong, no provider, still working...?

styx-tdo, 2022-05-02:
I have a Dell Latitude 5520
Dell Dock D6000, USB-C

The dock has 2x DP and 1x HDMI. I have 2 screens connected: 1x hdmi, 1x DP. I have signal only on the HDMI screen.
 xrandr, however, tells me:
_

> xrandr |grep connec
> eDP-1 connected 1920x1080+1920+0 (normal left inverted right x axis y axis) 344mm x 194mm panning 1920x1080+1920+0
> HDMI-1 disconnected (normal left inverted right x axis y axis)
> DP-1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 527mm x 296mm panning 1920x1080+0+0
> DP-2 disconnected (normal left inverted right x axis y axis)

_

When plugging/unplugging a DP screen, something happens (screens get black and get restored), but the screens don't show up as connected.

Any idea on how to make this dock work with 2x DP would be appreciated.

The debug info:

> --------------- Linux system info ----------------
> 
> Distro: Debian
> Release: sid
> Kernel: 5.13.12-1-siduction-amd64
> 
> ---------------- DisplayLink info ----------------
> 
> Driver version: 
> DisplayLink service status: 
> EVDI service version: /sys/devices/evdi/version not found
> 
> ------------------ Graphics card -----------------
> 
> Vendor: i915
> Subsystem: [Iris
> VGA: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] (rev 01)
> VGA (3D): 
> X11 version: 21.1.3-2+b1
> X11 configs: /etc/X11/xorg.conf.d/00-keyboard.conf /etc/X11/xorg.conf.d/99-libinput.conf
> 
> -------------- DisplayLink xorg.conf -------------
> 
>
>File: /etc/X11/xorg.conf.d/20-displaylink.conf
>cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
>Contents:
> 
>
> -------------------- Monitors --------------------
> 
> Providers: number : 1
> Provider 0: id: 0x46 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 4 outputs: 4 associated providers: 0 name:modesetting
> 
> -------------------------------------------------------------------

styx-tdo, 2022-05-02:
Additional info: Added a 2nd Dock (D1000), that doesn't show anything at all.
I start thinking that the dock just passes through the HDMI port the notebook does have actually, and that evdi doesn't do anything at all. Will downgrade to 1.9, who knows...

Barabazs, 2022-05-29:
From your debug info, I see that displaylink is not installed.

closed 2022-07-08 by Barabazs