AdnanHodzic/displaylink-debian #714: Installation problem Kali Linux on Raspberry

tomcdev63, 2022-05-04:
Hi, I'm trying to install an AOC E1659FWU monitor that only connects to USB3 on my raspberry. But like davidmans7 I have a problem with the installation...

[image]

apt search linux-source
apt show linux-source

[image]

Barabazs, 2022-05-04:
Is there a displaylink chip in your raspberry pi?

tomcdev63, 2022-05-04:
I don't know, it's an original raspberry pi 4b, I didn't modify anything

Barabazs, 2022-05-04:
Do you intend to connect your display directly to the Raspberry Pi (with usb)?
Then you will need to know if there is a DisplayLink chip in the RPI. (I don't think there is one) 

If there is no chip, then you will need a DisplayLink adapter to connect the display to the RPI.

tomcdev63, 2022-05-04:
It has already worked! Thanks to these drivers (https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu) I got an image. But when I rebooted my KaliLinux everything crashed.
I had to reinstall Kali... Apparently these drivers are designed for Ubuntu and not KaliLinux

tomcdev63, 2022-05-05:
I think it's the same problem as davidmans7... not a material problem

closed 2022-05-29 by Barabazs