AdnanHodzic/displaylink-debian #715: Ran displaylink-debian.sh, now only a black screen.

cypheroxide, 2022-05-08:
Running a multiboot Legion 5 laptop.  Ran the installer as per the instructions, rebooted, and got to a login screen.  After login, black screen with no video output.  Tried plugging in a monitor on different ports, still nothing.  Restarted several times, still get a black screen after login.  Using one of my other installed systems to write this.  Navigating to the _/etc/X11/xorg.conf.d_ directory to look at the file _20-displaylink.conf_, this is the output:

`    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

`

I'm running an AMD Ryzen 5 4600h and Nvidia GTX 1650Ti.  What can I do to get my normal display back?  Forget trying to get multiple monitors working like I have set up on Ubuntu Studio or Windows, I just want my Kali install back up and running again.  If I can at least get back to a terminal, I'd be fine with that.  I have a Timeshift backup just before running this thing.  Before being asked, 1. I confirmed my distro was on the compatibility list, 2. I ran the script and rebooted as per the instruction, which immediately lead to this issue, 3. I have no display output anywhere and cannot get back into my system to fix anything.

cypheroxide, 2022-05-08:
Nevermind.  Decided, _on a whim_, to save the .conf file to a backup file and deleted the original.  After a reboot, I'm back in.  Now that I know how to restore my display out, I'm going to try to diagnose and will return with my findings.  If I figure it out, this will be closed.

closed 2022-05-08 by cypheroxide

edgarbjorntvedt, 2024-09-15:
Thank you very much. This helped me a lot. Got black screen **before** login screen.
  Lenovo Thinkpad T14 Gen 5
  Popos 22.04
Entered console on popos with Ctrl+Alt+F5

"solution": 
```
cd  /etc/X11/xorg.conf.d/
mv 20-displaylink.conf 20-displaylink.conf.backup
```
reboot