AdnanHodzic/displaylink-debian #716: Have you tried with installing either the nvidia-lts or nivida-beta drivers?

arturogatti, 2022-05-10:
Have you tried with installing either the nvidia-lts or nivida-beta drivers?
That seems to be a frequent solution.

_Originally posted by @Barabazs in https://github.com/AdnanHodzic/displaylink-debian/issues/710#issuecomment-1109378312_

arturogatti, 2022-05-10:
yes. it is still a black screen, but it lets me set the providers. 

--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-10-amd64

---------------- DisplayLink info ----------------

Driver version: 1.10.0
1.10.1
DisplayLink service status: up and running
EVDI service version: 1.10.1

------------------ Graphics card -----------------

Vendor: nvidia
Subsystem: [NVS
GT218
VGA: NVIDIA Corporation GF119 [NVS 310] (rev a1)
NVIDIA Corporation GT218 [GeForce 210] (rev a2)
VGA (3D):
X11 version: 1.20.11-1+deb11u1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
Section "ServerLayout"
Identifier "layout"
Screen 0 "nvidia"
Inactive "intel"
EndSection

Section "Device"
Identifier "intel"
Driver "modesetting"
Option "AccelMethod" "None"
EndSection

Section "Screen"
Identifier "intel"
Device "intel"
EndSection

Section "Device"
Identifier "nvidia"
Driver "nvidia"
Option "ConstrainCursor" "off"
EndSection

Section "Screen"
Identifier "nvidia"
Device "nvidia"
Option "AllowEmptyInitialConfiguration" "on"
Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x279 cap: 0x1, Source Output crtcs: 2 outputs: 4 associated providers: 4 name:NVIDIA-0
Provider 1: id: 0x31c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x2fa cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x2d8 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x2ab cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

Barabazs, 2022-05-29:
Please don't open a new issue for the same problem. 
Duplicate of #710

closed 2022-05-29 by Barabazs