AdnanHodzic/displaylink-debian #717: EVDI fail install

RubenVerster, 2022-05-14:
Hello

I'm having issues with installing the evdi dependancy for these community drivers for DisplayLink

```
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j16 KERNELRELEASE=5.13.0-41-generic all INCLUDEDIR=/lib/modules/5.13.0-41-generic/build/include KVERSION=5.13.0-41-generic DKMS_BUILD=1...(bad exit status: 2)
ERROR (dkms apport): binary package for evdi: 1.10.1 not found
Error! Bad return status for module build on kernel: 5.13.0-41-generic (x86_64)
Consult /var/lib/dkms/evdi/1.10.1/build/make.log for more information.

------------------------------
Deleting module version: 1.10.1
completely from the DKMS tree.
------------------------------
Done.
ERROR: Failed to install evdi to the kernel tree.
```

Every time I try to install evdi, I have to rebuild my kernel
I'm on Ubuntu 20.05 and kernel 15.3

output from --debug

```
--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.13.0-41-generic

---------------- DisplayLink info ----------------

Driver version: 5.1.26
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Ltd.
VGA: Intel Corporation Device 9a60 (rev 01)
VGA (3D): NVIDIA Corporation Device 25a5 (rev a1)
X11 version: 1.20.13-1ubuntu1~20.04.2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x45 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 2 associated providers: 0 name:modesetting

```

Barabazs, 2022-05-29:
How did you try to install evdi?

RubenVerster, 2022-05-29:
@Barabazs Through dpkg (if I remember correctly) when I try to install DisplayLink drivers from the official package that DisplayLink provides, I get errors at the EVDI section. Also when I install the community version, I get errors at thte EVDI installs
If I try to install EVDI, I have to rebuild my kernel everytime, since it interrupts other installs as well :page_with_curl:

Barabazs, 2022-06-01:
And what do you get when you try to install with this script?

RubenVerster, 2022-06-01:
The first terminal output I initially posted is from installing the community version by following the documentation they provide

RubenVerster, 2022-06-01:
I will try to install again later today after work :)

RubenVerster, 2022-06-01:
I installed it now again and it looks successful
I installed updates for Ubuntu a while back, so I think they fixed all this now :)

```
--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.13.0-46-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.synaptics.com/node/4601?filetype=exe
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2022-06-01 15:58:53--  https://www.synaptics.com/sites/default/files/exe_files/2022-05/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu5.6-EXE.zip
Resolving www.synaptics.com (www.synaptics.com)... 54.245.106.105
Connecting to www.synaptics.com (www.synaptics.com)|54.245.106.105|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 16906259 (16M) [application/zip]
Saving to: ‘DisplayLink_Ubuntu_5.6.zip’

DisplayLink_Ubuntu_5.6.zip                           100%[=====================================================================================================================>]  16.12M  83.4KB/s    in 1m 53s  

2022-06-01 16:00:48 (146 KB/s) - ‘DisplayLink_Ubuntu_5.6.zip’ saved [16906259/16906259]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.6.zip
  inflating: 5.6/DisplayLink USB Graphics Software for Ubuntu 5.6.0-59.176-Release Notes.txt  
  inflating: 5.6/displaylink-driver-5.6.0-59.176.run  
Creating directory displaylink-driver-5.6.0-59.176
Verifying archive integrity...  100%   MD5 checksums are OK. All good.
Uncompressing DisplayLink Linux Driver 5.6.0-59.176  100%  
patching file 5.6/displaylink-driver-5.6/displaylink-installer.sh
patch unexpectedly ends in middle of line
Hunk #1 succeeded at 526 (offset 24 lines).

Installing driver version: 5.6

Distribution discovered: Ubuntu 20.04.4 LTS

Installing
[ Installing EVDI ]
[[ Installing EVDI DKMS module ]]

Creating symlink /var/lib/dkms/evdi/1.11.0/source ->
                 /usr/src/evdi-1.11.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j16 KERNELRELEASE=5.13.0-46-generic all INCLUDEDIR=/lib/modules/5.13.0-46-generic/build/include KVERSION=5.13.0-46-generic DKMS_BUILD=1...
Signing module:
 - /var/lib/dkms/evdi/1.11.0/5.13.0-46-generic/x86_64/module/evdi.ko
Secure Boot not enabled on this system.
cleaning build area...

DKMS: build completed.

evdi.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.13.0-46-generic/updates/dkms/

depmod....

DKMS: install completed.
[[ Installing module configuration files ]]
[[ Installing EVDI library ]]
make: Entering directory '/tmp/tmp.haCVMsIvEd/evdi/library'
cc -I../module -std=gnu99 -fPIC -D_FILE_OFFSET_BITS=64    -c -o evdi_lib.o evdi_lib.c
cc evdi_lib.o -shared -Wl,-soname,libevdi.so.0 -o libevdi.so.1.11.0 -lc -lgcc 
cp libevdi.so.1.11.0 libevdi.so
make: Leaving directory '/tmp/tmp.haCVMsIvEd/evdi/library'
[ Installing x64-ubuntu-1604/DisplayLinkManager ]
[ Installing libraries ]
[ Installing firmware packages ]
[ Installing licence file ]
[ Adding udev rule for DisplayLink DL-3xxx/4xxx/5xxx/6xxx devices ]
[ Adding upstart and powermanager sctripts ]

Please read the FAQ
http://support.displaylink.com/knowledgebase/topics/103927-troubleshooting-ubuntu
Adding udlfb to blacklist
Adding udl to blacklist

-------------------------------------------------------------------

Performing post install steps

Enable displaylink-driver service
Created symlink /etc/systemd/system/multi-user.target.wants/displaylink-driver.service → /lib/systemd/system/displaylink-driver.service.
Setup DisplayLink xorg.conf depending on graphics card
/usr/share/sddm/scripts/Xsetup not found, creating
Wrote changes to /usr/share/sddm/scripts/Xsetup
Wrote X11 changes to: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------------------------------------------------------------

Performing clean-up
Removing redundant: "DisplayLink_Ubuntu_5.6.zip" file
Removing redundant: "5.6" directory

-------------------------------------------------------------------

Install complete, please reboot to apply the changes
```

Barabazs, 2022-06-02:
> I've waited a while for this XD

I'm glad it worked out for you!