AdnanHodzic/displaylink-debian #718: EVDI version not found error

tupkovic, 2022-05-17:
When installing DisplayLink 5.5 for Ubuntu 22.04, install goes okay but evdi doesnt appear to select a version for DisplayLink? See below debug with error `/sys/devices/evdi/version not found`:

`------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: jammy
Kernel: 5.15.39-051539-generic

---------------- DisplayLink info ----------------

Driver version: 1.10.1
DisplayLink service status: 
**EVDI service version: /sys/devices/evdi/version not found**

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [UHD
VGA: Intel Corporation WhiskeyLake-U GT2 [UHD Graphics 620] (rev 02)
VGA (3D): 
X11 version: 21.1.3-2ubuntu2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 1 name:modesetting
Provider 1: id: 0xe8 cap: 0x4, Source Offload crtcs: 5 outputs: 0 associated providers: 1 name:modesetting

`

Barabazs, 2022-05-29:
How did you install displaylink?

tupkovic, 2022-05-29:
> How did you install displaylink?

I installed it viathe instructions from this github, and tried using synaps direct drivers. I had to remove evdi-dkms from my system completely,and than used the latest synaps drivers from their website which target Ubuntu 22.04. Though I think it was just the evdi-dkms removal that would have fixed the issue as no matter what I did (including having installed and built evdi, multiple versions attempted) it would always result in evdi not found error above. Ive now finally got it working after weeks of fiddling around so this can probably be closed.

closed 2022-05-30 by Barabazs