AdnanHodzic/displaylink-debian #729: Request - add support for PureOS

Smojo, 2022-07-24:
As PureOS (https://pureos.net/) is
- based on Debian 11 (Bullseye)
- Linux Kernel 5.10 LTS 
- Windowing System: Wayland
- Gnome 3.38.5

So I hope adding the support is not to complicated. Just donated 25€. I know that is no guarantee, but in any case it will hopefully help you. Thanks for your effort!

Let me know if you need more information of my system.

Debug-Info:
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: PureOS
Release: byzantium
Kernel: 5.10.0-16-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
ath9k
Subsystem: GT2
AR9462
VGA: Intel Corporation Skylake GT2 [HD Graphics 520] (rev 07)
VGA (3D): 
X11 version: 1.20.11-1+deb11u1
X11 configs: /etc/X11/xorg.conf.d/00-keyboard.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 0

-------------------------------------------------------------------

Smojo, 2022-07-24:
As I found PureOS already in (displaylink-debian.sh) there, I just added 
`|| [ $codename == "byzantium" ]` to line 291. 

Now the script stops with:
------------------------------------------
reparing for install

Archive:  DisplayLink_Ubuntu_5.6.zip
  inflating: 5.6/DisplayLink USB Graphics Software for Ubuntu 5.6.0-59.176-Release Notes.txt  
  inflating: 5.6/displaylink-driver-5.6.0-59.176.run  
Creating directory displaylink-driver-5.6.0-59.176
Verifying archive integrity...  100%   MD5 checksums are OK. All good.
Uncompressing DisplayLink Linux Driver 5.6.0-59.176  100%  
./displaylink-debian.sh: line 478: /tmp/resources/displaylink-installer.patch: No such file or directory
-----------------------------

Dependencies are fine btw.:
------------------------------
Checking dependencies

unzip is installed
linux-headers-5.10.0-16-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...
------------------------------

Smojo, 2022-07-27:
Ok, solved on my own.

_**Background:**_
I have i-tec docking station (bought it, because of the promoted quite good linux support)

**_What was the mistake?_** 
In the package there is a little quick-setup instruction and you can also navigate on the vendors homepage to get the driver for linux. This guided me to this
https://i-tec.cz/wp-content/uploads/man_dri/DisplayLink_Ubuntu.zip
In this Zip there are two instruction documents (others are for Ubuntu):

1. For other Linux distirbutions.docx
2. For other Linux distirbutions.pdf

And even this was not stupid enough (also from my site), both describe different things.
I went for method 2 of the pdf: 
```
Download and run script without Git (method 2)

wget https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/displaylink-debian.sh
chmod +x displaylink-debian.sh && sudo ./displaylink-debian.sh
```
I don't know why. Maybe I though for a short moment, that it is quicker/smaller to just pull the script via wget. 

Anyway today I followed the instruction you guys described on the main page of the repo:
https://github.com/AdnanHodzic/displaylink-debian#how-to-run-displaylink-debian 

--> et voilà the missing resources folder is part of the clone and after doing the change to the script I described in my last post, it ran through like charm.

Both Displays turned on directly and I can smoothly adjust which one is the primary or which one should be turned of via Gnome Display settings. Didn't expect that it works soo good.
Thanks again for all you work on this.

Lessons learned follow the authors instruction and not some random vendor you just points to it and created his own.

Maybe in an upcoming PR / new release you add 
|| [ $codename == "byzantium" ] to line 291.

I can confirm it works fine.

closed 2022-07-27 by Smojo

Barabazs, 2022-08-05:
Thank you for the feedback @Smojo.
I'll make a PR with your suggestion :pray: