AdnanHodzic/displaylink-debian #730: Debian 11 Kernel 5.10.0-16-amd64 EVDI service version not found

sarura2010, 2022-07-28:
HI , 

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-16-amd64

---------------- DisplayLink info ----------------

Driver version: 1.10.0
1.10.1
1.11.0
1.9.1
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
intel_ish_ipc
Subsystem: 620
Integrated
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.20.11-1+deb11u1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

Help please . 

Thank you

Barabazs, 2022-08-05:
What happens when you try the install procedure?

MorganeF06, 2022-08-09:
Hi, I have the same problem : 
sudo ./displaylink-debian.sh --install

Output :
--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.10.0-16-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.synaptics.com/node/4601?filetype=exe
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2022-08-09 09:24:15--  https://www.synaptics.com/sites/default/files/exe_files/2022-05/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu5.6-EXE.zip
Résolution de www.synaptics.com (www.synaptics.com)… 54.245.106.105
Connexion à www.synaptics.com (www.synaptics.com)|54.245.106.105|:443… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 16906259 (16M) [application/zip]
Sauvegarde en : « DisplayLink_Ubuntu_5.6.zip »

DisplayLink_Ubuntu_ 100%[===================>]  16,12M  8,48MB/s    ds 1,9s    

2022-08-09 09:24:18 (8,48 MB/s) — « DisplayLink_Ubuntu_5.6.zip » sauvegardé [16906259/16906259]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.6.zip
  inflating: 5.6/DisplayLink USB Graphics Software for Ubuntu 5.6.0-59.176-Release Notes.txt  
  inflating: 5.6/displaylink-driver-5.6.0-59.176.run  
Creating directory displaylink-driver-5.6.0-59.176
Verifying archive integrity...  100%   MD5 checksums are OK. All good.
Uncompressing DisplayLink Linux Driver 5.6.0-59.176  100%  
patching file 5.6/displaylink-driver-5.6/displaylink-installer.sh
patch unexpectedly ends in middle of line
Hunk #1 succeeded at 526 (offset 24 lines).

Installing driver version: 5.6

Distribution discovered: Debian GNU/Linux 11 (bullseye)

Installing
[ Installing EVDI ]
[[ Installing EVDI DKMS module ]]

Creating symlink /var/lib/dkms/evdi/1.11.0/source ->
                 /usr/src/evdi-1.11.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j16 KERNELRELEASE=5.10.0-16-amd64 all INCLUDEDIR=/lib/modules/5.10.0-16-amd64/build/include KVERSION=5.10.0-16-amd64 DKMS_BUILD=1...
cleaning build area...

DKMS: build completed.

evdi.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.0-16-amd64/updates/dkms/

depmod...

DKMS: install completed.
[[ Installing module configuration files ]]
[[ Installing EVDI library ]]
make: Entering directory '/tmp/tmp.8gsuR6lXUL/evdi/library'
cc -I../module -std=gnu99 -fPIC -D_FILE_OFFSET_BITS=64    -c -o evdi_lib.o evdi_lib.c
cc evdi_lib.o -shared -Wl,-soname,libevdi.so.0 -o libevdi.so.1.11.0 -lc -lgcc 
cp libevdi.so.1.11.0 libevdi.so
make: Leaving directory '/tmp/tmp.8gsuR6lXUL/evdi/library'
[ Installing x64-ubuntu-1604/DisplayLinkManager ]
[ Installing libraries ]
[ Installing firmware packages ]
[ Installing licence file ]
[ Adding udev rule for DisplayLink DL-3xxx/4xxx/5xxx/6xxx devices ]
[ Adding upstart and powermanager sctripts ]

Please read the FAQ
http://support.displaylink.com/knowledgebase/topics/103927-troubleshooting-ubuntu
Adding udlfb to blacklist
Adding udl to blacklist

-------------------------------------------------------------------

Performing post install steps

Enable displaylink-driver service
Created symlink /etc/systemd/system/multi-user.target.wants/displaylink-driver.service → /lib/systemd/system/displaylink-driver.service.
Setup DisplayLink xorg.conf depending on graphics card
Wrote X11 changes to: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------------------------------------------------------------

Performing clean-up
Removing redundant: "DisplayLink_Ubuntu_5.6.zip" file
Removing redundant: "5.6" directory

-------------------------------------------------------------------

Install complete, please reboot to apply the changes
After reboot, make sure to consult post-install guide! http://bit.ly/2TbZleK
Reboot now? [Y/n] y

MorganeF06, 2022-08-09:
I reboot my computer, then :

cd displaylink-debian/ && sudo ./displaylink-debian.sh --debug

Output :

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-16-amd64

---------------- DisplayLink info ----------------

Driver version: 1.11.0
1.9.0+dfsg
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 
VGA: Intel Corporation Device 9a60 (rev 01)
VGA (3D): NVIDIA Corporation Device 1fba (rev a1)
X11 version: 1.20.11-1+deb11u2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xrdp/xorg.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

[... 137 lines cut ...]

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

I need help please, thank you :)

Barabazs, 2022-08-12:
@MorganeF06 Please don't post your problem/logs in someone else's issue. I have created a separate issue for you in #737