AdnanHodzic/displaylink-debian #731: Installation issue on Ubuntu 22.04 lts

StormTrooper720, 2022-07-28:
When running `sudo ./displaylink-debian.sh` it waits for ~15 seconds then outputs `Can't open display`. I added the `--debug` argument but it returns the same thing. I am not that experienced using a linux terminal, so I don't know how I would fix this. Thanks for the help in advance.

Barabazs, 2022-08-05:
Can you give the output of these commands please?

`cat /sys/devices/evdi/version)`
`lsb_release -is`
`lsb_release -cs`
`lsb_release -ics | sed '$!s/$/ /' | tr -d '\n'`
`uname -r`
`lspci -nnk | grep -i vga -A3 | grep 'in use' | cut -d ':' -f2 | sed 's/ //g'`
`lspci -nnk | grep -i vga -A3 | grep Subsystem | cut -d ' ' -f5`
`dpkg -l | grep "ii  xserver-xorg-core" | awk '{print $3}' | sed 's/[^,:]*://g'`

StormTrooper720, 2022-08-07:
`cat /sys/devices/evdi/version`  -  `no such file or directory`
`lsb_release -is`  -  `Pop`
`lsb_release -cs`  -  `jammy`
`lsb_release -ics | sed '$!s/$/ /' | tr -d '\n'`  -  `Pop jammy`
`uname -r`  -  `5.18.10-76051810-generic`
`lspci -nnk | grep -i vga -A3 | grep 'in use' | cut -d ':' -f2 | sed 's/ //g'`  -  `nvidia`
`lspci -nnk | grep -i vga -A3 | grep Subsystem | cut -d ' ' -f5`  -  `Ltd.`
`dpkg -l | grep "ii xserver-xorg-core" | awk '{print $3}' | sed 's/[^,:]*://g'`  -  no output

Barabazs, 2022-08-12:
Apparently there is an upstream bug with kernel 5.18.xx
You can either try with an older kernel version or wait for a new evdi release... (unfortunately, out of our hands)

closed 2022-08-12 by Barabazs