AdnanHodzic/displaylink-debian #735: Please add support for devuan Chimaera

wfdudley, 2022-08-07:
Your script supports only Devuan ASCII which is very old.
I'm willing to contribute $$ for the effort to fix this.

Also, your script installs a LOT of software before deciding that
I'm on an unsupported platform.  Couldn't that check be done
up front so as to avoid installing source etc. if not used?

Thanks,
William F. Dudley

Barabazs, 2022-08-08:
Hi @wfdudley, can you post the output of `sudo ./displaylink-debian.sh --debug` please? 
That should give me the information needed to add the requested platfrom.

I'll look into your second question. It's a valid concern.

wfdudley, 2022-08-08:
Thanks for the prompt reply.

Please find that debug output below.

The following message was also sent to stdout:

"./displaylink-debian.sh: line 369: /etc/init.d/displaylink.sh: No such
file or directory
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory"

I note that the script  did NOT complain about unsupported version when I
ran it
this time.

Note that this is a fresh install of Devuan Chimaera on an Intel i7 machine.

Thanks,
Bill Dudley

------------------- output of --debug ---------------------------------
-------------------------------------------------------------------

Starting Debug ...

--------------- Linux system info ----------------

Distro: Devuan
Release: chimaera
Kernel: 5.10.0-16-amd64

---------------- DisplayLink info ----------------

Driver version:
DisplayLink service status:
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: E3-1200
VGA: Intel Corporation Xeon E3-1200 v2/3rd Gen Core processor Graphics
Controller (rev 09)
VGA (3D):
X11 version: 1.20.11-1+deb11u2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x45 cap: 0xf, Source Output, Sink Output, Source Offload,
Sink Offload crtcs: 3 outputs: 3 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
-------------------- end of --debug output -----------------------

wfdudley, 2022-08-09:
I can give you a login to a Devuan Chimaera machine with DisplayLink
installed,
and I can give you root access if you like, should you need a development
machine.

Bill Dudley
This email is free of malware because I run Linux.

Barabazs, 2022-08-12:
This release has now been added to supported platforms in #736 
Could you try to pull the latest changes and retry please?

wfdudley, 2022-08-30:
Why is this completed?  The script still fails to install the driver.

Thanks,
Bill Dudley
This email is free of malware because I run Linux.

wfdudley, 2022-10-11:
I did a git clone to get latest, and ran as root.

Result was this:

Preparing for install

Archive:  DisplayLink_Ubuntu_5.6.1.zip
  inflating: 5.6.1/displaylink-driver-5.6.1-59.184.run
  inflating: 5.6.1/DisplayLink USB Graphics Software for Ubuntu
5.6.1-59.185-Release Notes.txt
  inflating: 5.6.1/__MACOSX/._DisplayLink USB Graphics Software for Ubuntu
5.6.1-59.185-Release Notes.txt
chmod: cannot access '5.6.1/displaylink-driver-5.6.1.*[0-9]*-[0-9]*.run':
No such file or directory

The file that the chmod is looking for is this:
displaylink-driver-5.6.0-59.176.run

So I changed the chmod line (and the two following) to this:
chmod +x $driver_dir/displaylink-driver-${version}[-.0-9]*.run
./$driver_dir/displaylink-driver-${version}[-.0-9]*.run --keep --noexec
mv displaylink-driver-${version}[-.0-9]*/
$driver_dir/displaylink-driver-${version}

This solved the "chmod: cannot access" problem.  Script is still failing:

Preparing for install

Archive:  DisplayLink_Ubuntu_5.6.1.zip
  inflating: 5.6.1/displaylink-driver-5.6.1-59.184.run
  inflating: 5.6.1/DisplayLink USB Graphics Software for Ubuntu
5.6.1-59.185-Release Notes.txt
  inflating: 5.6.1/__MACOSX/._DisplayLink USB Graphics Software for Ubuntu
5.6.1-59.185-Release Notes.txt
Creating directory displaylink-driver-5.6.1-59.184
Verifying archive integrity...  100%   MD5 checksums are OK. All good.
Uncompressing DisplayLink Linux Driver 5.6.1-59.184  100%
patching file 5.6.1/displaylink-driver-5.6.1/displaylink-installer.sh
patch unexpectedly ends in middle of line
Hunk #1 succeeded at 526 (offset 24 lines).

Installing driver version: 5.6.1

ERROR: the installer script is unable to find out how to start
DisplayLinkManager service automatically on your system.
Please set an environment variable SYSTEMINITDAEMON to 'upstart', 'systemd'
or 'runit' before running the installation script to force one of the
options.
Installation terminated.

output of displaylink-debian.sh --debug below:

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x
[y/N] y

--------------- Linux system info ----------------

Distro: Devuan
Release: chimaera
Kernel: 5.10.0-16-amd64

---------------- DisplayLink info ----------------

Driver version:
./displaylink-debian.sh: line 369: /etc/init.d/displaylink.sh: No such file
or directory
DisplayLink service status:
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: E3-1200
VGA: Intel Corporation Xeon E3-1200 v2/3rd Gen Core processor Graphics
Controller (rev 09)
VGA (3D):
X11 version: 1.20.11-1+deb11u2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x45 cap: 0xf, Source Output, Sink Output, Source Offload,
Sink Offload crtcs: 3 outputs: 3 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

Bill Dudley
This email is free of malware because I run Linux.