AdnanHodzic/displaylink-debian #738: Install fails with version 5.6.1 of the driver.zip due to incorrect glob

ajfclark, 2022-08-12:
```
$ sudo ./displaylink-debian.sh 

--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.18.0-2-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.synaptics.com/node/4781?filetype=exe
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2022-08-13 08:25:15--  https://www.synaptics.com/sites/default/files/exe_files/2022-08/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu5.6.1-EXE.zip
Resolving www.synaptics.com (www.synaptics.com)... 54.245.106.105
Connecting to www.synaptics.com (www.synaptics.com)|54.245.106.105|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 16905857 (16M) [application/zip]
Saving to: ‘DisplayLink_Ubuntu_5.6.1.zip’

DisplayLink_Ubuntu_5.6.1.zip                         100%[=====================================================================================================================>]  16.12M  5.60MB/s    in 2.9s    

2022-08-13 08:25:18 (5.60 MB/s) - ‘DisplayLink_Ubuntu_5.6.1.zip’ saved [16905857/16905857]

Removing prior: "5.6.1" directory

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.6.1.zip
  inflating: 5.6.1/displaylink-driver-5.6.1-59.184.run  
  inflating: 5.6.1/DisplayLink USB Graphics Software for Ubuntu 5.6.1-59.185-Release Notes.txt  
  inflating: 5.6.1/__MACOSX/._DisplayLink USB Graphics Software for Ubuntu 5.6.1-59.185-Release Notes.txt  
chmod: cannot access '5.6.1/displaylink-driver-5.6.1.*[0-9]*-[0-9]*.run': No such file or directory

```

dkebler, 2022-08-15:
same with me.  Looks like this PR was not added to master :(.

ajfclark, 2022-08-15:
@dkebler I only put it up a couple of days ago. You can pull it from my repo in the meantime: https://github.com/ajfclark/displaylink-debian/tree/issue-738

dkebler, 2022-08-15:
@AdnanHodzic  Just reporting this branch fixed the issue for me so Ok? to merge.  Thx @ajfclark

commit 461ad757 (2022-08-15): Fixes #738
closed 2022-08-15 by Barabazs