AdnanHodzic/displaylink-debian #742: How to update?

Smojo, 2022-08-21:
I was looking for a recommended way to update the driver, but I did not really find find something in the [post install](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md), [troubleshooting](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#troubleshooting-most-common-issues) or the main [README.md](https://github.com/AdnanHodzic/displaylink-debian/blob/master/README.md). 

I was also looking for an issue about it but e.g. this one #495 was just closed due to in-activity.

I was thinking about this when I saw that a newer Ubuntu driver was release at (at this point when I'm opening this issue -> release 5.6.1 | 20220810) https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu 
I installed the driver via displaylink-debian at 20220727 and it downloads and uses (modifies/grabs the importent files) of the latest Ubuntu driver during the execution, so **I thought it might make sense to update the driver** and potentially fix some little hick-ups / issues.

So the questions is: 
#### What is the recommended way to perfom such a driver update/upgrade?

I didn't find an option to choose "update/upgrade". 

> Options:
> 
> [I]nstall
> [D]ebug
> [R]e-install
> [U]ninstall
> [Q]uit

I ended up in using re-install, but in my case it just uninstalled the current driver (confirmed that everything was removed and recommended to reboot) and I executed install after a reboot -> which worked fine ;-). So no big problem (if you confirm that this is the way to go, or maybe document it somewhere), but in between the two displays were off of course, what I felt is not 100% optimal.

varac, 2022-08-30:
I found https://github.com/AdnanHodzic/displaylink-debian/pull/655 related to this

Barabazs, 2023-03-04:
Please refer to https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#upgrading-your-os-or-updating-displaylink

closed 2023-03-04 by Barabazs