AdnanHodzic/displaylink-debian #745: Unsuported kernel version: 5.19

varac, 2022-08-30:
I try to reinstall/upgrade but this is what I got:

```
❯ sudo ./displaylink-debian.sh --debug

---------------------------------------------------------------

Unsuported kernel version: 5.19
Please wait for a stable DisplayLink release or try to install the beta
No support will be given for the beta version

This tool is Open Source and feel free to extend it
GitHub repo: https://github.com/AdnanHodzic/displaylink-debian/

---------------------------------------------------------------
```

I'm using PopOS 22.04, which is on the list of supported versions in the README.md.
Does the the upstream driver need an earlier kernel version, or is it this project ?

ricklambrechts, 2022-08-31:
Same, also for me on debian

calebanthony, 2022-08-31:
Got this working! I'm also on PopOS 22.04.

The issue I had was needing the newest version of EVDI. Maybe with these steps someone smarter than me can help get this into the repo.

1. Get the official DisplayLink installer here (https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu). I got v5.6.1.
2. Make sure your existing DisplayLink stuff is uninstalled. You should be able to do `displaylink-installer uninstall`.
3. Reboot. This is because EVDI can't be modified while you're in X.
4. Get the newest version of EVDI (I got 1.12.0) from their repo (https://github.com/DisplayLink/evdi/releases). Just download the source code.
5. Move the contents of `evdi-1.12.0/module` to `/usr/src/evdi-1.12.0`.
6. Run `dkms install evdi/1.12.0`. This will install the proper version of EVDI.
7. Run the DisplayLink installer you grabbed in Step 1 and run the `*.run` file. You should see no errors.
8. Reboot and enjoy!

As a side note, I normally have to unplug my dock while booting and plug it back in after boot in order to get things working properly, but now I no longer have to do that. Not sure if that's recent or if I've been doing my superstitious voodoo unnecessarily for a while.

godlewski, 2022-09-04:
Having the same issue on Ubuntu 22.04.1 LTS

varac, 2022-09-06:
@calebanthony Thanks for your detailled instructions. I'll rather wait for a proper fix in displaylink-debian though, hoping it will soon land.

PR #750 (2022-09-13): Add kinetic and 5.19 support
Add support for `kinetic`, which is due out in October.
`kinetic` will be using the 5.19 kernel, so makes sense to get that added

The output below shows that 1.12.0 of `evdi` works on 5.19, so should be all good

```
$ dkms status
evdi/1.12.0, 5.15.0-46-generic, x86_64: installed
evdi/1.12.0, 5.19.0-15-generic, x86_64: installed
```

Closes #745
closed 2022-09-13 by Barabazs