AdnanHodzic/displaylink-debian #746: BunsenLabs Lithium support not working

jimjamz, 2022-08-30:
I can confirm that the installer is not working correctly for BunsenLabs Lithium.  I pulled the latest from master on 2022-08-23.
The following are terminal commands and my responses to help diagnose the issue:
`uname -r`:

`4.19.0-21-amd64`

`lsb_release -a`:
```
Distributor ID:	Bunsenlabs
Description:	BunsenLabs GNU/Linux 10.5 (Lithium)
Release:	10.5
Codename:	buster
```

`lsb_release -is`:

`Bunsenlabs`

`lsb_release -cs`:

`buster`

When running the _displaylink-debian.sh_ script, I receive the following:
```
---------------------------------------------------------------

Unsuported platform: Bunsenlabs buster
Full list of all supported platforms: http://bit.ly/2zrwz2u

```

Which is strange because `$lsb` variable is assigned to `$(lsb_release -is)`, `$codename` variable is assigned to `lsb_release -cs` and from line 259 both conditions are met.

Barabazs, 2022-09-13:
I merged a fix for your issue. Can you try again with the latest version and report back please?

closed 2022-09-13 by Barabazs

jimjamz, 2022-10-09:
The installation issue has been fixed.

PR #751 (2022-09-13): fix: bunsenlabs string match
Closes  #746