AdnanHodzic/displaylink-debian #747: not working under Xorg under Ubuntu 22.04

user706, 2022-08-31:
Hi,

the display link is **working nicely** with my HP docking station (HP USB-C/A Universal dock G2) and HP notebook (HP Elitebook 645 G9), but only **when using Wayland** (have 2 monitors connected to docking station).

When connected **via Xorg, nothing seems to be working**.

--> Is that normal??

`lsb_release -a` shows
```
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy
```

`echo -e 'y\ny\n' | sudo ./displaylink-debian.sh --debug` shows:
```
-------------------------------------------------------------------

Starting Debug ...

--------------- Linux system info ----------------

Distro: Ubuntu
Release: jammy
Kernel: 5.15.0-46-generic

---------------- DisplayLink info ----------------

Driver version: 1.12.0
1.9.1
DisplayLink service status: up and running
EVDI service version: 1.12.0

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: [103c:89d2]
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Barcelo (rev c1)
VGA (3D): 
X11 version: 21.1.3-2ubuntu2.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier "AMDGPU"
    Driver     "amdgpu"
    Option     "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x56 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 4 associated providers: 0 name:Unknown AMD Radeon GPU @ pci:0000:05:00.0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

amenk, 2022-09-13:
same here, wayland was working fine.
My Login screen still works, when logging-in to xorg, the screens switch off

arif-ali, 2022-09-14:
Seems like I had the same issue, when I upgraded to the new displaylink-debian, and the fact it created new `/etc/X11/xorg.conf.d/20-displaylink.conf`. I renamed this file to `20-displaylink.conf.bak`, and then logout and log in again.

This, I believe will do the right thing, and automatically

amenk, 2022-09-14:
I tried removing the `20-displaylink.conf` but it did not seem to help in may case. I have secure boot on though, but the debug output looks like everything is properly loaded.

arif-ali, 2022-09-14:
ah, interesting, I don't have secure boot, so could be something :thinking:

closed 2023-06-04 by Barabazs