AdnanHodzic/displaylink-debian #748: USB-C dock is switching off while Laptop is connected

kybernetik42, 2022-08-31:
Hi,
HP G6 USB-C Docking station
HP Probook 650 G5-C port. 
I start the laptop via USB-C Dock. Two external screens are switich on, I can logon, but then the USB-C Dock is switching off =:-/
When I power recycle the USC-C Dock, then I can switch on again, but the it take a while and the the USB-C Dock is switiching off again.
I assume it that the Laptop is switching off the USB.

I cannot really judge, what is going on.
First I paste the debug text, which I get, after the USB-C Dock has switched off!
 below I put a lot of lines from dmesg.

--------------- Linux system info ----------------

Distro: Linuxmint
Release: una
Kernel: 5.4.0-125-generic

---------------- DisplayLink info ----------------

Driver version: 1.12.0
1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation UHD Graphics 620 (Whiskey Lake) (rev 02)
VGA (3D): 
X11 version: 1.20.13-1ubuntu1~20.04.3
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 8 associated providers: 4 name:modesetting
Provider 1: id: 0x13c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x11a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xf8 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xd6 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

   26.769871] usb 1-5: new high-speed USB device number 4 using xhci_hcd
[   27.610790] usb 1-5: New USB device found, idVendor=03f0, idProduct=036b, bcdDevice= 6.11
[   27.610795] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   27.610798] usb 1-5: Product: HP USB-C Dock G5
[   27.610801] usb 1-5: Manufacturer: Microchip
[   27.611790] hub 1-5:1.0: USB hub found
[   27.611827] hub 1-5:1.0: 5 ports detected
[   27.781916] usb 2-5: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[   27.802240] usb 2-5: New USB device found, idVendor=03f0, idProduct=066b, bcdDevice= 6.11
[   27.802246] usb 2-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   27.802249] usb 2-5: Product: HP USB-C Dock G5
[   27.802251] usb 2-5: Manufacturer: Microchip
[   27.803392] hub 2-5:1.0: USB hub found
[   27.803425] hub 2-5:1.0: 4 ports detected
[   27.897827] usb 1-5.2: new high-speed USB device number 5 using xhci_hcd
[   28.567547] usb 1-5.2: New USB device found, idVendor=03f0, idProduct=056b, bcdDevice= 0.06
[   28.567559] usb 1-5.2: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[... 213 lines cut ...]
[  142.123278] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  142.123292] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  142.264182] usb 2-5: USB disconnect, device number 5
[  142.264184] usb 2-5.3: USB disconnect, device number 6
[  142.264731] usb 2-5.4: USB disconnect, device number 7
[  144.002206] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  144.002228] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  144.002243] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  144.002256] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  144.572650] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  144.572669] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  144.572681] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  144.572692] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  152.330878] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  152.330885] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  152.330921] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  152.330926] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  152.330958] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  152.330963] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  152.330994] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  152.330999] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  152.374431] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  152.374445] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  152.374452] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected
[  152.374461] evdi: [D] evdi_detect:94 (dev=0) poll connector state: disconnected

Barabazs, 2022-12-27:
It's not clear what the issue might be here. I would suggest filing a bug on the DisplayLink forum.

closed 2022-12-27 by Barabazs

kybernetik42, 2022-12-27:
Oh, sorry, I should answer before: It seems that the USB-C dock is 
broken :-( DisplayLink is working fine. I still will for the replacement.