AdnanHodzic/displaylink-debian #753: Bunsenlabs Lithium - Providers present but displays not recognised

jimjamz, 2022-10-09:
After installing, I initially had only 1 provider when running `sudo ./displaylink-debian.sh --debug`.
After following both troubleshooting guides for Intel display adapters, I now have 5 providers (replacing either [this](https://github.com/AdnanHodzic/displaylink-debian/issues/228#issuecomment-467889348) X11 entry or completely removing _/etc/X11/xorg.conf.d/20-displaylink.conf_ produce the same result):

```
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Bunsenlabs
Release: buster
Kernel: 4.19.0-22-amd64

---------------- DisplayLink info ----------------

Driver version: 1.12.0
DisplayLink service status: up and running
EVDI service version: 1.12.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [HD
VGA: Intel Corporation HD Graphics 520 (rev 07)
VGA (3D): 
X11 version: 1.20.4-1+deb10u5
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 # Section "Device"
#     Identifier  "Intel"
#     Driver      "intel"
# EndSection

Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0xcf cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

`xrandr --listproviders`:
```
Providers: number : 5
Provider 0: id: 0xcf cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
```

However, `xrandr -q` shows any additional displays as disconnected:
```
Screen 0: minimum 320 x 200, current 1366 x 768, maximum 8192 x 8192
eDP-1 connected primary 1366x768+0+0 (normal left inverted right x axis y axis) 276mm x 155mm
   1366x768      60.00*+  48.00  
   1360x768      59.80    59.96  
   1280x720      60.00    59.99    59.86    59.74  
   1024x768      60.04    60.00  
   960x720       60.00  
   928x696       60.05  
   896x672       60.01  
   1024x576      59.95    59.96    59.90    59.82  
   960x600       59.93    60.00  
   960x540       59.96    59.99    59.63    59.82  
   800x600       60.00    60.32    56.25  
   840x525       60.01    59.88  
   864x486       59.92    59.57  
   800x512       60.17  
   700x525       59.98  
   800x450       59.95    59.82  
   640x512       60.02  
   720x450       59.89  
   700x450       59.96    59.88  
   640x480       60.00    59.94  
   720x405       59.51    58.99  
   684x384       59.88    59.85  
   680x384       59.80    59.96  
   640x400       59.88    59.98  
   576x432       60.06  
   640x360       59.86    59.83    59.84    59.32  
   512x384       60.00  
   512x288       60.00    59.92  
   480x270       59.63    59.82  
   400x300       60.32    56.34  
   432x243       59.92    59.57  
   320x240       60.05  
   360x202       59.51    59.13  
   320x180       59.84    59.32  
DP-1 disconnected (normal left inverted right x axis y axis)
HDMI-1 disconnected (normal left inverted right x axis y axis)
DP-2 disconnected (normal left inverted right x axis y axis)
HDMI-2 disconnected (normal left inverted right x axis y axis)
```
My HDMI monitor connected to the USB-C dock is not recognised.
When the HDMI cable is removed from the dock, and connected directly to my ThinkPad, HDMI-2 is recognised:
```
HDMI-2 connected (normal left inverted right x axis y axis)
   1920x1080     60.00 +  50.00    59.94  
   1920x1080i    60.00    50.00    59.94  
   1680x1050     59.88  
   1600x900      60.00  
   1280x1024     75.02    60.02  
   1280x800      59.91  
   1152x864      75.00  
   1280x720      60.00    50.00    59.94  
   1024x768      75.03    60.00  
   832x624       74.55  
   800x600       75.00    60.32  
   720x576       50.00  
   720x576i      50.00  
   720x480       60.00    59.94  
   720x480i      60.00    59.94  
   640x480       75.00    60.00    59.94  
   720x400       70.08 
```

Secure Boot has been disabled in the BIOS.

Barabazs, 2022-12-27:
Did you associate the providers as described in the [post-install guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#display-detection)?

jimjamz, 2023-01-14:
No, I hadn't.  Associating the providers solved the issue.
Unfortunately, this is the end of my use of the driver as I remembered that DisplayLink does not support any blue-light reduction apps for redshift, f.lux, Night light, on either Linux or MacOS,

jimjamz, 2023-01-14:
Author provided the solution to the problem.

closed 2023-01-14 by jimjamz