AdnanHodzic/displaylink-debian #755: Framework Laptop with Ubuntu 22.04.1 LTS dont detects screens on icy box

KaySchneider, 2022-10-24:
Im using the ICY-Box: IB-DK2251AC     
 

Debug Output 

````
--------------- Linux system info ----------------

Distro: Ubuntu
Release: jammy
Kernel: 5.15.0-52-generic

---------------- DisplayLink info ----------------

Driver version: 1.12.0
DisplayLink service status: up and running
EVDI service version: 1.12.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 
VGA: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
VGA (3D): 
X11 version: 21.1.3-2ubuntu2.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x49 cap: 0x9, Source Output, Sink Offload crtcs: 5 outputs: 6 associated providers: 0 name:Intel

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
````

closed 2022-10-24 by KaySchneider

Jarvvski, 2022-12-12:
You say when you used wayland it worked?

I'm running a framework laptop too - display link doesn't seem to detect any monitors for me

KaySchneider, 2023-01-18:
hi, yes. With wayland the icy box can detect the screens.