AdnanHodzic/displaylink-debian #756: after installation reboot system goes in terminal mode

sjovanovic91, 2022-10-25:
Hello,

for first I would connect a display with a usb to hdmi adapter and I use Buster 10 with Kernel 4.19.0-22-amd64.

After installation and reboot, my Debian Buster boots in terminal mode. I cant start the X11 -Server, the error: "no screen founds". And the `sudo ./displaylink-debian.sh --debug` shows me the error "Cant open Display".

After searching I found the solution to remove the   `/etc/X11/xorg.conf.d/20-displaylink.conf`. Than my system boot normaly in KDE GUI.

After the display detection with `xrandr --setprovideroutputsource <prov-xid> <source-xid>` and the the command `sudo ./displaylink-debian.sh --debug` show my this:

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-22-amd64

---------------- DisplayLink info ----------------

Driver version: 1.12.0
DisplayLink service status: up and running
EVDI service version: 1.12.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Core
VGA: Intel Corporation 4th Generation Core Processor Family Integrated Graphics Controller (rev 06)
VGA (3D): 
X11 version: 1.20.4-1+deb10u5

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0xcf cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 4 name:modesetting
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

I think my system detected the displays but the screen with the usb to hdmi adapter have no signal. The reason is maybe because now the File: `/etc/X11/xorg.conf.d/20-displaylink.conf `is not found .. 

I hope someone has a solution.

Thanks

Barabazs, 2022-12-27:
There must be a configuration issue in the `/etc/X11/xorg.conf.d/20-displaylink.conf` file?
Can you post the content please.

codename-cn, 2023-01-06:
Same issue

content of `/etc/X11/xorg.conf.d/20-displaylink.conf`:

```
Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection
```

smithseaz, 2023-01-14:
Dropping this here as update and for myself as a future reference, and for anyone that may be having some trouble with getting the driver installed and running.  Much of this is in the documents and post install guides, this is just a centralization of that data.

I am having the same issue.  I have  a fresh install of Debian11 (Kernel 5.0.10-20-amd64) on a Lenovo P53s.

After running the initial install script and prompted reboot my system hangs at starting display link drivers.  Looking at sjovanovic91 post about removing the 20-displaylink.conf file to enable their system to boot correctly. I booted into recovery mode and  made a copy of the file as requested by  Barabazs and have attached it here, then removed it from the /etc/X11/Xorg.conf.d path and rebooted successfully.  After reboot I noted that the driver was not loading (secure boot needed to be disabled), rebooted into BIOS and disabled.  Now the driver loads, I have all three monitors being identified by xrandr.

There is a note buried in the original docs that says renaming the 20-displaylink-conf file may be a good fix, in my case that was 100% accurate and the file does not appear to be needed.  I would be curious to see what you (Barbazs) see's in the file that may make it work as expected when put back into the path.

Adnan, Thank you for getting this setup and keeping it current!  I have been using this driver for the past couple of years with great success (generally after a little digging--- but that is part of the fun right?)

Cheers!

[20-displaylink.conf.txt](https://github.com/AdnanHodzic/displaylink-debian/files/10418444/20-displaylink.conf.txt)