AdnanHodzic/displaylink-debian #757: Unsuported kernel version: 6.0

varac, 2022-10-29:
PopOS 22.04 LTS just upgraded to kernel `linux-image-6.0.2-76060002-generic` and now I can't use [displaylink-debian](https://github.com/AdnanHodzic/displaylink-debian) anymore:

```
❯ sudo ./displaylink-debian.sh --uninstall

---------------------------------------------------------------

Unsuported kernel version: 6.0
Please wait for a stable DisplayLink release or try to install the beta
No support will be given for the beta version

This tool is Open Source and feel free to extend it
GitHub repo: https://github.com/AdnanHodzic/displaylink-debian/

---------------------------------------------------------------

```

varac, 2022-10-29:
Related: #745

surgiie, 2022-11-07:
Looking for this support. :cry:

surgiie, 2022-11-07:
FWIW, i was able to get this to work with this method ( https://www.displaylink.org/forum/showpost.php?p=94504&postcount=8):

```bash
cd /tmp
git clone git@github.com:DisplayLink/evdi.git
cd evdi
make
```

List all installed versions of evdi: ("evdi/<version>" is what you are looking for)
```bash
dkms status
```
Uninstall current version of evdi (if the latest version (currently 1.12.0 is not installed at all, this step is unnecessary)), if you want you can uninstall earlier versions, too

```bash
sudo dkms remove -m evdi -v 1.12.0 --all
```
Create the dkms source folder and copy the newly built files into the dkms folder:
```bash
sudo mkdir /usr/src/evdi-1.12.0/
cd /usr/src/evdi-1.12.0/
sudo cp /tmp/evdi/module/* .

```
Register the dkms module:
```bash
sudo dkms add -m evdi -v 1.12.0
sudo dkms build -m evdi -v 1.12.0
sudo dkms install -m evdi -v 1.12.0
```
If after a reboot, it does not just start working, re-run ubuntu installer (from and following the instructions [https://www.synaptics.com/products/d...wnloads/ubuntu](https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu)):

```bash
sudo ./displaylink-driver-5.6.1-59.184.run
```

Timmphy, 2022-11-12:
@surgiie thanks for these steps. 
But if use this with fresh installed Debian 12 (testing) with Kernel 6.0.0.3 it dont work. 
At first glance it seems to be working. In the login screen you are in a loop. The laptop screen goes off every second. Black screen 1s <-> login screen 1s. When I unplug the USB Dock, the system runs normally. But it can be seen that in the loop the other screens are trying to start.

surgiie, 2022-11-12:
@Timmphy Unfortunate to hear, I should of mentioned, im running Pop OS 22.04. Its possible some steps may differ on Debian. :/

GGyll, 2022-11-14:
`> FWIW, i was able to get this to work with this method on Pop OS 22.04 ( https://www.displaylink.org/forum/showpost.php?p=94504&postcount=8):
`
Weird, I am running Pop OS as well and my docking station HDMI screen stopped working after the update. it displays the gray background during login but after that there is no signal. Followed your steps but am still getting

```
Unsuported kernel version: 6.0
Please wait for a stable DisplayLink release or try to install the beta
No support will be given for the beta version
```

surgiie, 2022-11-15:
> `> FWIW, i was able to get this to work with this method on Pop OS 22.04 ( https://www.displaylink.org/forum/showpost.php?p=94504&postcount=8): ` Weird, I am running Pop OS as well and my docking [...]

It appears like you are still trying to run the installer via this repository install script after following the steps. You need to download the official installer and run that

https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu

NaNaDi, 2022-11-22:
I get the same message about unsupported kernel version 6.0 on Debian 11. Hoping for an update soon.

jean-emmanuel, 2022-11-25:
This worked for me : https://code.berrydejager.com/Fix-DisplayLink_drivers-linux-kernel-6/

NaNaDi, 2022-12-08:
> This worked for me (debian testing / linux 6.0) : https://code.berrydejager.com/Fix-DisplayLink_drivers-linux-kernel-6/

Did not work for me on Debian 11. The installation worked, but the second screen is still black.

jezh42, 2022-12-14:
I am running Debian Bookworm/Testing with kernel 6.0.0 and none of the above solutions were working for me. However, I was able to get it working by modifying the displaylink-debian.sh script to accomodate for the previously mentioned fix (https://code.berrydejager.com/Fix-DisplayLink_drivers-linux-kernel-6/).

My fix involved changing the `max_kernel_version_supported` to `6.0.0` on line 20 and adding the following two lines on ~463 before `# get sysinitdaemon`

`curl -L https://github.com/DisplayLink/evdi/archive/refs/heads/devel.tar.gz -o $driver_dir/displaylink-driver-${version}/evdi.tar.gz
sed -i 's#if ! tar xf "$TARGZ" -C "$EVDI"; then#if ! tar xf "$TARGZ" -C "$EVDI" --strip-components=1; then#g' $driver_dir/displaylink-driver-${version}/displaylink-installer.sh`

renoproc, 2022-12-16:
> This worked for me (debian testing / linux 6.0) : https://code.berrydejager.com/Fix-DisplayLink_drivers-linux-kernel-6/

Solution tested with a new installed Kali 6, worked like a charm !

eremt, 2023-01-14:
> This worked for me (debian testing / linux 6.0) : https://code.berrydejager.com/Fix-DisplayLink_drivers-linux-kernel-6/

Can confirm, also on debian testing with kernel 6.0.

Barabazs, 2023-01-15:
Thank you all for the comments and suggestions.
I have now updated the script in #775.

closed 2023-01-15 by Barabazs

geekdenz, 2023-07-30:
Downloading https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu and running the installer with `sudo sh <downloaded-file>` where `<downloaded-file` is the `*.run` file worked for me.