AdnanHodzic/displaylink-debian #762: dock ultra - no displays shown

twarnick, 2022-11-15:
Have here a Lenovo X1 Yoga (1 gen.) with ThinkPad Ultra Dock.

-------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-18-amd64

---------------- DisplayLink info ----------------

Driver version: 1.12.0
DisplayLink service status: up and running
EVDI service version: 1.12.0

------------------ Graphics card -----------------

Vendor: i915
intel_ish_ipc
Subsystem: [HD
Integrated
VGA: Intel Corporation Skylake GT2 [HD Graphics 520] (rev 07)
VGA (3D): 
X11 version: 1.20.11-1+deb11u2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf.d/20-intel.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x49 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 6 associated providers: 0 name:Intel

Barabazs, 2022-11-26:
Are you sure that is a DisplayLink compatible device?

twarnick, 2022-11-28:
How can I verifying that the device is compatible?

Barabazs, 2022-11-29:
Look at the specifications or contact the seller.

Barabazs, 2022-12-27:
Closing this issue due to inactivity.

Please reopen this issue if it still persists.

closed 2022-12-27 by Barabazs

twarnick, 2023-01-25:
Hi,
in /var/displaylink/ are files that has the name of my  monitors (DELL_xyxz)
so, I think, the device is working correct?