AdnanHodzic/displaylink-debian #764: The computer does not start if the dock is plugged in at boot time

neonicoo, 2022-11-27:
--------------- Linux system info ----------------

Distro: Linux Mint 21
Release: based on ubuntu 22.04 jammy
Kernel: 5.15.0-53-generic

---------------USB C DOCK  info ----------------

Lenovo Thinkpad Hybryd usb-c dock station

---------------- DisplayLink info ----------------

DisplayLink : 5.6.1
EVDI service version: 1.12.0

------------------ Graphics card -----------------

dGPU : NVIDIA T1200 Laptop GPU
Igpu : Intel TigerLake-H GT1
NVIDIA driver : 520.56.06

-------------------- Monitors --------------------
3 screens : 1 integrated screen + 2 external monitors

Providers: number : 6
Provider 0: id: 0x217 cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0x317 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x2f6 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x2d5 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x2b4 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x240 cap: 0x2, Sink Output crtcs: 4 outputs: 6 associated providers: 1 name:modesetting

----------------- Issue ---------------------------

When I turn on my computer (HP zbook power G8) and the dock is connected in usb-c to it, the computer remains stuck on the linux mint logo and does not start. I am obliged to connect my dock only once the computer has correctly started.

Note that I have the same problem with my second computer (lenovo legion 5), under linux mint 21 with the same kernel 5.15.

erzhan89, 2022-11-29:
on my Thinkpad X1 Extreme gen 2. I had same issue.. 
Fix was to Disable security of Thunderbolt 3 in BIOS.

neonicoo, 2022-12-03:
Hi @erzhan89 , thanks for your answer.
I have the same issue with my Lenovo legion 5 15ach6h which is not thunderbold 3 compatible ...

neonicoo, 2022-12-03:
EDITO: in fact my computer starts, but after a very long time, more or less 10min with the dock connected, against 5-10s without the dock.

Barabazs, 2022-12-27:
It's a known issue, but unfortunately without a solution for now... 

> EDITO: in fact my computer starts, but after a very long time, more or less 10min with the dock connected, against 5-10s without the dock.

 
> I also noticed that neither redshift nor any other temperature control application works on the external displays connected via displaylink

This is also a known issue. Upstream support is needed, but very unlikely to ever happen...
https://github.com/DisplayLink/evdi/issues/44

closed 2022-12-27 by Barabazs