AdnanHodzic/displaylink-debian #765: Feature request: SysV init support

bybor, 2022-12-04:
Would you please consider adding SysV support? Thank you!

Barabazs, 2022-12-27:
This is out-of-scope for this repo. Do you have a specific problem where you need this?

bybor, 2022-12-27:
Yes, sure, I understand.

I'm using live environment that doesn't have systemd (mx linux), which means that installer stops with an error.

It will be great if installer could copy all required files, but leave it to user - how to start the service. I see why you may not like this feature - users will come and open issues about their displays "not working".

Thank you for your work and time you devote to it. It's obviously a very niche thing (sysV).  Happy holidays!

I think it should be possible to do a manual installation based on the files that your release, i don't have enough experience yet to do it, maybe one day.

closed 2024-05-03 by Barabazs