AdnanHodzic/displaylink-debian #766: Unsuported kernel version: 6.0

bill-fischer, 2022-12-04:
Hello. I was excited to find this project since the drivers offered by DisplayLink do not support the current PopOS kernel. However, it appears this project does not work with the newer kernel either. Please see below:

===================================
:~/displaylink-debian$ sudo ./displaylink-debian.sh --debug

---------------------------------------------------------------

Unsuported kernel version: 6.0
Please wait for a stable DisplayLink release or try to install the beta
No support will be given for the beta version

This tool is Open Source and feel free to extend it
GitHub repo: https://github.com/AdnanHodzic/displaylink-debian/
===================================

Would you be able to update your project to support the latest driver version and kernel? I would gladly make a donation.

Thank you!

faysalnadim, 2022-12-05:
hi Bill, i'm facing the similar challenge here. please post here if you find a solution. i checked on the official display link support page as well and it seem to be not supported either. 

https://www.synaptics.com/products/displaylink-graphics/downloads/macos

faysalnadim, 2022-12-05:
I just reloaded back to Debian 5.19 using recovery feature for the time being. would suggest the same to anyone who faces similar challenge until an official patch is received.

tddpirate, 2022-12-05:
Please see also: issue https://github.com/AdnanHodzic/displaylink-debian/issues/757

faysalnadim, 2022-12-10:
> Please see also: issue #757

greatly appreciated, i was able to fix this issue using the solution in issue #757

foreverLoveWisdom, 2022-12-14:
Hi guys, running into the same issue on POP OS 22.04, anybody figure out a hack around this? Tried the solution in 757, but does not work for me

Barabazs, 2022-12-27:
Duplicate of #757

closed 2022-12-27 by Barabazs