AdnanHodzic/displaylink-debian #767: Monitor Portátil Asus ZenScreen 15.6 Full HD, IPS, USB Type-C, - MB16ACE

rjcfj, 2022-12-10:
--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-19-amd64

---------------- DisplayLink info ----------------

Driver version: 1.12.0
DisplayLink service status: up and running
EVDI service version: 1.12.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation Iris Plus Graphics G1 (Ice Lake) (rev 07)
VGA (3D): 
X11 version: 1.20.11-1+deb11u3
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x46 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 0 name:Intel

timemrah, 2022-12-25:
--------------- Linux system info ----------------

Distro: Deepin
Release: apricot
Kernel: 5.15.77-amd64-desktop

---------------- DisplayLink info ----------------

Driver version: 1.12.0
DisplayLink service status: up and running
EVDI service version: 1.12.0

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: [Radeon
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Picasso (rev c1)
VGA (3D): 
X11 version: 1.20.11-1+deb11u1
X11 configs: /etc/X11/imwheel/startup.conf /etc/X11/xorg.conf.d/50-synaptics.conf /etc/X11/xorg.conf.d/05-x100.conf /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf.d/75-wacom.conf /etc/X11/xorg.conf.d/10-smi.conf /etc/X11/xorg.conf.d/40-libinput.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier "AMDGPU"
    Driver     "amdgpu"
    Option     "PageFlip" "false"
EndSection

Same Monitor

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x57 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 4 outputs: 4 associated providers: 0 name:AMD Radeon Vega 10 Graphics @ pci:0000:05:00.0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

Barabazs, 2022-12-27:
What is your issue?

timemrah, 2022-12-27:
When I plug in the monitor it doesn't work.

Barabazs, 2022-12-27:
Do you connect it with USB-A?

timemrah, 2022-12-27:
USB A - 3.1 - i connect to  thinkpad E595

[image]

Barabazs, 2022-12-27:
And the debug output you posted above is with the display connected and turned on?
What does `lsusb` output?

timemrah, 2022-12-27:
Unfortunately I can't give you what you want. Because I tried a bunch of linux distros to run my external monitor. Finally the driver works in Popos 20.04 operating system. I'm currently using Popos 20.04.

Barabazs, 2022-12-27:
> Unfortunately I can't give you what you want. Because I tried a bunch of linux distros to run my external monitor. Finally the driver works in Popos 20.04 operating system. I'm currently using Popos 20.04.

Why did you post in this issue if you don't have the same problem?

timemrah, 2022-12-28:
While using Deepin 20.8 and 20.3, I wrote that my screen does not work with the same monitor as in this topic. I'm sorry if I made a mistake. I thought maybe it would be of use to you.

rjcfj, 2022-12-29:
Hello, several attempts to install drivers and restart linux, maybe linux something dirty or poorly installed. I formatted debian 11 latest version and installed the displaylink driver and restarted it got the monitor image everything perfect with using it with USB-A that basically, Thanks

[image]

timemrah, 2022-12-29:
I had no problems with USB-A when I installed the official displaylink driver on Debian 11.6.0 GNOME. But when I installed the driver here it didn't work.

rjcfj, 2022-12-29:
> Não tive problemas com USB-A quando instalei o driver displaylink oficial no Debian 11.6.0 GNOME. Mas quando instalei o driver aqui não funcionou.

rjcfj, 2022-12-29:
Linux version 5.10.0-20-amd64 (debian-kernel@lists.debian.org) (gcc-10 (Debian 10.2.1-6) 10.2.1 20210110, GNU ld (GNU Binutils for Debian) 2.35.2) #1 SMP Debian 5.10.158-2 (2022-12-13)

closed 2022-12-29 by rjcfj