AdnanHodzic/displaylink-debian #769: duex plus monitor + displaylink

autumncheney, 2022-12-26:
`sudo ./displaylink-debian.sh --debug`:
```
--------------- Linux system info ----------------

Distro: Debian
Release: bookworm
Kernel: 6.0.0-6-amd64

---------------- DisplayLink info ----------------

Driver version: 1.12.0
1.12.0
1.12.0+dfsg
DisplayLink service status: up and running
EVDI service version: 1.12.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: GT2
[GeForce
VGA: Intel Corporation CometLake-H GT2 [UHD Graphics] (rev 05)
NVIDIA Corporation TU117M [GeForce GTX 1650 Ti Mobile] (rev a1)
VGA (3D): 
X11 version: 21.1.5-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf.d/20-evdi.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 6 associated providers: 4 name:modesetting
Provider 1: id: 0x15d cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x13c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x11b cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xfa cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------
```

`lsusb`:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 04f2:b685 Chicony Electronics Co., Ltd Chicony USB2.0 Camera
Bus 001 Device 003: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 005: ID 0489:e076 Foxconn / Hon Hai 
Bus 001 Device 007: ID 0711:560b Magic Control Technology Corp. T6 USB Station
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
trying to get my duex plus monitor to work with my system (system76 gazelle, sid). i had to modify the `displaylink-debian.sh` file to support my 6.0.0.6 kernel, hopefully i didn't mess anything up in the process

`lsusb` tells me the monitor is detected by the system (it's the `Magic Control Technology Corp. T6 USB Station`), but the monitor itself just says there is no signal and goes into power saving mode

is there anything i can do here?

Barabazs, 2022-12-27:
Are you sure it's a DisplayLink device?

autumncheney, 2022-12-27:
i contacted the manufacturer about that, still waiting for a response

autumncheney, 2022-12-28:
alright, so they got back to me

no, it does not use the displaylink protocol. but, at least in my case, i can buy a cable that connects to the displayport mini port on my laptop and get it to work that way

sorry for all the trouble :(

closed 2022-12-28 by autumncheney