AdnanHodzic/displaylink-debian #772: displaylink won't show external displays

zero-pytagoras, 2023-01-10:
Hi,
I am using Debian and I am trying to use displaylink, and it seems to fail. below are the details of  system

--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-20-amd64

---------------- DisplayLink info ----------------

Driver version: 1.12.0
DisplayLink service status: up and running
EVDI service version: 1.12.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation HD Graphics 620 (rev 02)
VGA (3D): 
X11 version: 1.20.11-1+deb11u3
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf.d/00-keyboard.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting
Provider 1: id: 0x13b cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x119 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0xf7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0xd5 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

mylo87, 2023-01-14:
Hello. 
I am an amateur.
I can not get by this stage in post install guide; command not found

Make sure that UEFI/secure boot is disabled!

When you are logging into your session you are using X server (X11) and not Wayland.

After install you must reboot to apply the changes.

After reboot, make sure DispayLink is running by running debug i.e: **sudo ./displaylink-debian.sh --debug**

sudo ./displaylink-debian.sh --debug
sudo: ./displaylink-debian.sh: command not found
sudo ./displaylink-debian.sh --debug
sudo: ./displaylink-debian.sh: command not found

Thank you

Barabazs, 2023-01-15:
@zero-pytagoras 
You have to associate the providers, as noted in the [post-install-guide#display-detection](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#display-detection).

Barabazs, 2023-01-15:
@mylo87 Please open a new issue with all the required information.

closed 2023-06-04 by Barabazs