AdnanHodzic/displaylink-debian #773: Unsupported Mint Vera

riccardonar, 2023-01-12:
Do you can add the support for Mint Vera?

```bash
-------------- Linux system info ----------------

Distro: Linuxmint
Release: vera
Kernel: 5.15.0-57-generic

---------------- DisplayLink info ----------------

Driver version: 1.12.0
DisplayLink service status: up and running
EVDI service version: 1.12.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: GT2
VGA: Intel Corporation WhiskeyLake-U GT2 [UHD Graphics 620]
VGA (3D): NVIDIA Corporation GM108M [GeForce MX110] (rev a2)
X11 version: 21.1.3-2ubuntu2.5

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: File o directory non esistente
Contents:
 

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x45 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 4 associated providers: 5 name:modesetting
Provider 1: id: 0x11f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xfe cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xdd cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xbc cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x9b cap: 0x4, Source Offload crtcs: 0 outputs: 0 associated providers: 1 name:modesetting
```

Barabazs, 2023-01-15:
Hi, I've added this version to the script.

closed 2023-01-15 by Barabazs

riccardonar, 2023-01-15:
> Hi, I've added this version to the script.

Wow. Thank you!

PR #774 (2023-01-15): feat: add support for Linux Mint Vera 
Closes #773