AdnanHodzic/displaylink-debian #778: Wayland support

janvlug, 2023-01-18:
It seems that displaylink-debian does not support Wayland, when reading the [prerequisites](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#prerequisites):

 “When you are logging into your session you are using X server (X11) and not Wayland.”

However Displaylink [should have Wayland support](https://github.com/DisplayLink/evdi/issues/58).

When I do: `sudo -E ./displaylink-debian.sh --debug` on my Librem 5 running PureOS, I get:

`Can't open display`

My guess is that this is because the phone runs Wayland.

See also [this forum topic](https://forums.puri.sm/t/displaylink-on-librem-5/19228).

janvlug, 2023-01-23:
I outcommented this line:
`providers="$(xrandr --listproviders)"`
the script continues now after this line.

janvlug, 2023-01-23:
Both on my Librem 5 with PureOS and on my Fedora 37 Wayland only systems `xrandr --listproviders` gives `Can't open display`. The command `xrandr --listmonitors` however gives output like:
```
Monitors: 1
 0: +WAYLAND0 360/65x720/130+0+0  XWAYLAND0
```

janvlug, 2023-02-27:
I tried installing displaylink-debian again on my Librem 5. After installation, the login screen does not appear, and the screen stays black. I noticed this in the prerequisites: "When you are logging into your session you are using X server (X11) and not Wayland." Maybe this is the reason why the login screen stayed black after the reboot.
Fortunately, I could still login via ssh to remove displaylink-debian. After uninstalling the login screen was there again after a reboot.

cyclops1982, 2023-12-04:
+1 on this issue. Debian bookworm comes standard with wayland. Would love to help out and make this work :)

zz0rk, 2024-05-22:
Please support Wayland. Thank you

vnd, 2024-06-11:
FWIW somewhat surprisingly it actually works for me on Wayland (Debian bookworm) and not X11 (heavily glitchy to the point of being unusable), AMD Ryzen 7000 laptop with integrated graphics.

4SH-gaupee, 2024-06-11:
Yeah same for me since evdi update, I only have a black screen on SDDM but the rest is working, I get a lot of CPU usage but this is a DisplayLink driver problem not the implementation of this script

vnd, 2024-06-12:
May be I was too quick to consider it fully working, so far the only issue is that after reboot I need to re-plug DL monitor into another USB port, then it's properly detected. Could be worse I guess, glad that it's working in principle.

Epsicle, 2024-08-31:
Just chiming in to say this is working with no modifications on Debian 12 Bookworm using KDE Plasma and Wayland. I initiall tried @janvlug's modifications but that failed outright on my system so I ran the script as is and it just worked.

My HP Elitebook G4 and Targus DOCK-190USZ-81 work great together now.

I think this line in the post-install guide could probably be removed based on mine and other experiences here. Or maybe changed to just a discouragement for wayland and encouragement for X11?
> When you are logging into your session you are using X server (X11) and not Wayland.

 To be honest, I completely disregarded this project when I found out about it a few months back because of this line, but it's working flawlessly on Wayland. Awesome work.

vnd, 2024-08-31:
I'm a bit puzzled, but I think on wayland it somehow works regardless of this project, as I'm typing this on DisplayLink monitor without even evdi running.

Epsicle, 2024-08-31:
> I'm a bit puzzled, but I think on wayland it somehow works regardless of this project, as I'm typing this on DisplayLink monitor without even evdi running.

If your using an AMD or Nvidia GPU that has displaylink ports, then I think your fine due to the drivers for those cards. But I don't think they're included in the iGPU drivers for some laptops and that causes issues with docking stations.

What are your system specs?

vnd, 2024-08-31:
> If your using an AMD or Nvidia GPU that has displaylink ports, then I think your fine due to the drivers for those cards. But I don't think they're included in the iGPU drivers for some laptops and [...]

AMD Ryzen 7000 laptop with integrated graphics, KDE Plasma

Epsicle, 2024-08-31:
> > > I'm a bit puzzled, but I think on wayland it somehow works regardless of this project, as I'm typing this on DisplayLink monitor without even evdi running.
> > 
> > 
> > If your using an AMD or Nvidia GPU that has displaylink ports, then I think your fine due to the drivers for those cards. But I don't think they're included in the iGPU drivers for some laptops and that causes issues with docking stations.
> > What are your system specs?
> 
> AMD Ryzen 7000 laptop (Thinkpad P16s) with integrated graphics, KDE Plasma

You might be covered by your AMD drivers then. My 7th gen intel does not work at all without this project.

Epsicle, 2024-08-31:
Oh wait, I'm dumb. My wayland login isnt working on the two displaylink displays, I'm just using my built in display to sign in.

Regardless I think the docs could be changed to indicate that so long as you don't care to using the displays your hooking up to login to your machine, they will work fine after you login.