AdnanHodzic/displaylink-debian #784: Unable to install DisplayLink on Kali 2022.4 kernel 6.0.0x

m-attique, 2023-01-31:
Hi,
I'm trying to install DisplayLink provided "**displaylink-debian.sh**" script. But unable to make it work, it gets stuck after reboot on blinking prompt.
**Debugged script output is as below. Please provide the support for this latest version of Kali kernel:**

**************************************** OUTPUT of Debug of script ***********************************************************
```
 # ./displaylink-debian.sh --debug      

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Kali
Release: kali-rolling
Kernel: 6.0.0-kali6-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation HD Graphics 530 (rev 06)
VGA (3D): NVIDIA Corporation GM107M [GeForce GTX 950M] (rev a2)
X11 version: 21.1.4-3

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x43 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 2 associated providers: 1 name:modesetting
Provider 1: id: 0xc5 cap: 0x5, Source Output, Source Offload crtcs: 0 outputs: 0 associated providers: 1 name:modesetting

-------------------------------------------------------------------

**************************************** OUTPUT of Debug of script ***********************************************************
```

=================================:::::

**Script ends with success and reboot options as shown below:**

=================================:::::

```

Performing post install steps

Enable displaylink-driver service
Created symlink /etc/systemd/system/multi-user.target.wants/displaylink-driver.service → /lib/systemd/system/displaylink-driver.service.
Setup DisplayLink xorg.conf depending on graphics card
/usr/share/sddm/scripts/Xsetup not found, creating
Wrote changes to /usr/share/sddm/scripts/Xsetup
Wrote X11 changes to: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------------------------------------------------------------

Performing clean-up
Removing redundant: "DisplayLink_Ubuntu_5.6.1.zip" file
Removing redundant: "5.6.1" directory

-------------------------------------------------------------------

Install complete, please reboot to apply the changes
After reboot, make sure to consult post-install guide! http://bit.ly/2TbZleK
Reboot now? [Y/n] 
```

=================================:::::

Thanks & Regards,
Attique

Barabazs, 2023-03-04:
Hi @m-attique it appears that there is a problem with installing evdi:
`EVDI service version: /sys/devices/evdi/version not found`

I would suggest retrying the install procedure and copy pasting the entire install terminal output here.

Barabazs, 2023-06-04:
Closing in favor of #828

closed 2023-06-04 by Barabazs