AdnanHodzic/displaylink-debian #785: displaylink issue with EVDI service version: /sys/devices/evdi/version not found

shendrenbit, 2023-01-31:
here is my debug information. i am trying to get 3 external monitors working on my thinkpad x1C. just bought an "i-tec" docking station but not recognizing monitors with the USB-C (or any other) video connection to my laptop. 

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Linuxmint
Release: vera
Kernel: 5.15.0-58-generic

---------------- DisplayLink info ----------------

Driver version: 1.12.0
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [Iris
VGA: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] (rev 01)
VGA (3D): 
X11 version: 21.1.3-2ubuntu2.5
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x4c cap: 0x9, Source Output, Sink Offload crtcs: 5 outputs: 9 associated providers: 0 name:Intel

shendrenbit, 2023-01-31:
also i have this.. 
~$ dkms status
evdi/1.12.0, 5.15.0-56-generic, x86_64: built
evdi/1.12.0, 5.15.0-57-generic, x86_64: installed
evdi/1.12.0, 5.15.0-58-generic, x86_64: installed

shendrenbit, 2023-02-01:
from doing some reading on this site i updated my /etc/X11/xorg.conf.d/20-displaylink.conf to look like

Section "OutputClass"
    Identifier         "DisplayLink"
    MatchDriver   "evdi"
    Driver              "modesetting"
    Option             "AccelMethod" "none"
EndSection

shendrenbit, 2023-02-02:
deleted the 20-displaylink.conf file and rebooted.. still no go... this is frustrating.

Barabazs, 2023-03-04:
Without evdi installed, you will not get a working setup.
How did you try to uninstall displaylink-debian?
Can you retry and post the full install terminal output?

shendrenbit, 2023-03-04:
your device does not work. if you think it will work please share the exact
instructions and command line entries for me to get it to work. i have 10
other engineers waiting for me to see what device they should buy.

*~~*
*Regards,*
*Scott Hendren*
*772-633-3180*
*Healthcare Systems Integration LLC*

Barabazs, 2023-03-08:
> your device does not work. if you think it will work please share the exact instructions and command line entries for me to get it to work. 
> 
Can you retry and post the full install terminal output?

> i have 10 other engineers waiting for me to see what device they should buy. 
> 
imo you shouldn't use this in a professional setting. Either use official displaylink installer and the supported Ubuntu version or upgrade your hardware to use thunderbolt.

closed 2023-06-04 by Barabazs