AdnanHodzic/displaylink-debian #789: Debian 11

dvinouze, 2023-02-10:
Hi,

This is the debug :

```
sudo ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-20-amd64

---------------- DisplayLink info ----------------

Driver version: 1.12.0
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: 
Subsystem: 
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Device 15e7 (rev c1)
VGA (3D): 
X11 version: 1.20.11-1+deb11u5
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```
Thanks

dvinouze, 2023-02-13:
#786

Barabazs, 2023-03-04:
Can you clarify what problem you are experiencing please?

iamdey, 2023-05-30:
This is probably due to an error when building evdi, see #821 
I think this should be closed as duplicate (sorry for the spam, but it seems to be duplicated several times)

closed 2023-06-04 by Barabazs