AdnanHodzic/displaylink-debian #797: Fail to install EVDI

lhambrecht, 2023-03-07:
After updating my system, my monitor were no longer detected.

Running`displaylink-debian.sh --debug` provided the following output:
```
---------------- DisplayLink info ----------------

Driver version: evdi
1.11.0
1.12.0
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found
```

An attempt to re-install displaylink failed with this output:
```
Preparing for install

Archive:  DisplayLink_Ubuntu_5.6.1.zip
  inflating: 5.6.1/displaylink-driver-5.6.1-59.184.run  
  inflating: 5.6.1/DisplayLink USB Graphics Software for Ubuntu 5.6.1-59.185-Release Notes.txt  
  inflating: 5.6.1/__MACOSX/._DisplayLink USB Graphics Software for Ubuntu 5.6.1-59.185-Release Notes.txt  
Creating directory displaylink-driver-5.6.1-59.184
Verifying archive integrity...  100%   MD5 checksums are OK. All good.
Uncompressing DisplayLink Linux Driver 5.6.1-59.184  100%  
patching file 5.6.1/displaylink-driver-5.6.1/displaylink-installer.sh
patch unexpectedly ends in middle of line
patching file 5.6.1/displaylink-driver-5.6.1/displaylink-installer.sh

Installing driver version: 5.6.1

Distribution discovered: Pop!_OS 22.04 LTS

Installing
[ Installing EVDI ]
patching file /tmp/tmp.ZXwpERiDfl/evdi/module/evdi_drm_drv.h
[[ Installing EVDI DKMS module ]]
Creating symlink /var/lib/dkms/evdi/1.12.0/source -> /usr/src/evdi-1.12.0

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=6.2.0-76060200-generic all INCLUDEDIR=/lib/modules/6.2.0-76060200-generic/build/include KVERSION=6.2.0-76060200-generic DKMS_BUILD=1...(bad exit status: 2)
ERROR (dkms apport): binary package for evdi: 1.12.0 not found
Error! Bad return status for module build on kernel: 6.2.0-76060200-generic (x86_64)
Consult /var/log/displaylink/evdi_install_make.2023-03-08-1033.log for more information.
Deleting module evdi-1.12.0 completely from the DKMS tree.
ERROR: Failed to install evdi to the kernel tree.
Adding udlfb to blacklist
Adding udl to blacklist

-------------------------------------------------------------------

Performing post install steps

sed: can't read /lib/systemd/system/displaylink-driver.service: No such file or directory
```

markvankirk, 2023-03-08:
I'm experiencing the same issue from a clean install.

Barabazs, 2023-03-08:
Kernel 6.2 is not supported yet. There are upstream issues.

Barabazs, 2023-03-08:
I just pushed a change that should fix the issue on kernel >6.2
Can you try again please?

lhambrecht, 2023-03-09:
Installation ran through without issues.
`displaylink-debian.sh --debug` output:

```
---------------- DisplayLink info ----------------

Driver version: evdi
1.11.0
1.12.0
DisplayLink service status: up and running
EVDI service version: 1.12.0
```

I am back at my workstation tomorrow and can check full functionality. But so far so good, thanks for the quick update.

lhambrecht, 2023-03-14:
Issue has been fixed and can be closed.

closed 2023-03-14 by lhambrecht