AdnanHodzic/displaylink-debian #799: Problem installing for POPOS 22.04

aron-lif-8157, 2023-03-08:
I get this error when running the install command.

[image]

Barabazs, 2023-03-08:
Please provide the evdi_install_make... log file that is mentioned at the end of your screenshot

maartenhendrix, 2023-03-09:
I have a similar issue.
```
--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-6.2.0-76060200-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.synaptics.com/node/4781?filetype=exe
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2023-03-09 10:33:25--  https://www.synaptics.com/sites/default/files/exe_files/2022-08/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu5.6.1-EXE.zip
Resolving www.synaptics.com (www.synaptics.com)... 54.245.106.105
Connecting to www.synaptics.com (www.synaptics.com)|54.245.106.105|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 16905857 (16M) [application/zip]
Saving to: ‘DisplayLink_Ubuntu_5.6.1.zip’

DisplayLink_Ubuntu_5.6.1.zip                         100%[=====================================================================================================================>]  16.12M  1.39MB/s    in 19s     

2023-03-09 10:33:45 (875 KB/s) - ‘DisplayLink_Ubuntu_5.6.1.zip’ saved [16905857/16905857]

Removing prior: "5.6.1" directory

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.6.1.zip
  inflating: 5.6.1/displaylink-driver-5.6.1-59.184.run  
  inflating: 5.6.1/DisplayLink USB Graphics Software for Ubuntu 5.6.1-59.185-Release Notes.txt  
  inflating: 5.6.1/__MACOSX/._DisplayLink USB Graphics Software for Ubuntu 5.6.1-59.185-Release Notes.txt  
Creating directory displaylink-driver-5.6.1-59.184
Verifying archive integrity...  100%   MD5 checksums are OK. All good.
Uncompressing DisplayLink Linux Driver 5.6.1-59.184  100%  
[... 345 lines cut ...]
  524 | #define __drm_dbg(fmt, ...)             ___drm_dbg(NULL, fmt, ##__VA_ARGS__)
      |                                                                 ^~~~~~~~~~~
/var/lib/dkms/evdi/1.12.0/build/evdi_fb.c:443:9: note: in expansion of macro ‘DRM_DEBUG_KMS’
  443 |         DRM_DEBUG_KMS("allocated %dx%d vmal %p\n",
      |         ^~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:252: /var/lib/dkms/evdi/1.12.0/build/evdi_painter.o] Error 1
/var/lib/dkms/evdi/1.12.0/build/evdi_fb.c: In function ‘evdi_fbdev_destroy’:
/var/lib/dkms/evdi/1.12.0/build/evdi_fb.c:481:17: error: implicit declaration of function ‘drm_framebuffer_unregister_private’ [-Werror=implicit-function-declaration]
  481 |                 drm_framebuffer_unregister_private(&efbdev->efb.base);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/evdi/1.12.0/build/evdi_fb.c: At top level:
/var/lib/dkms/evdi/1.12.0/build/evdi_fb.c:331:43: error: storage size of ‘evdifb_funcs’ isn’t known
  331 | static const struct drm_framebuffer_funcs evdifb_funcs = {
      |                                           ^~~~~~~~~~~~
/var/lib/dkms/evdi/1.12.0/build/evdi_fb.c: In function ‘evdi_framebuffer_init’:
/var/lib/dkms/evdi/1.12.0/build/evdi_fb.c:350:1: error: control reaches end of non-void function [-Werror=return-type]
  350 | }
      | ^
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:252: /var/lib/dkms/evdi/1.12.0/build/evdi_fb.o] Error 1
make[1]: *** [Makefile:2021: /var/lib/dkms/evdi/1.12.0/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.2.0-76060200-generic'
make: *** [Makefile:33: all] Error 2
```

jacksavage69, 2023-03-12:
I've been having the same problem

maartenhendrix, 2023-03-23:
https://github.com/DisplayLink/evdi/issues/406 might give a bit more insight.

maartenhendrix, 2023-04-20:
I see DisplayLink has just released version 5.7 and added a repository. 
https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu

maartenhendrix, 2023-04-20:
I see DisplayLink has just released version 5.7 and added a repository. 
https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu

For me it works without the script from this repo.

closed 2023-06-04 by Barabazs