AdnanHodzic/displaylink-debian #800: Installation issue on Kali (6.1.0-kali5-amd64)

sansquoi, 2023-03-08:
[Debug Info]:

--------------- Linux system info ----------------

Distro: Kali
Release: kali-rolling
Kernel: 6.1.0-kali5-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
intel_ish_ipc
Subsystem: 620
Integrated
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 21.1.7-1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x46 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting

-------------------------------------------------------------------

[Installation Messages]:

Preparing for install

Archive:  DisplayLink_Ubuntu_5.6.1.zip
  inflating: 5.6.1/displaylink-driver-5.6.1-59.184.run  
  inflating: 5.6.1/DisplayLink USB Graphics Software for Ubuntu 5.6.1-59.185-Release Notes.txt  
  inflating: 5.6.1/__MACOSX/._DisplayLink USB Graphics Software for Ubuntu 5.6.1-59.185-Release Notes.txt  
Creating directory displaylink-driver-5.6.1-59.184
Verifying archive integrity...  100%   MD5 checksums are OK. All good.
Uncompressing DisplayLink Linux Driver 5.6.1-59.184  100%  
patching file 5.6.1/displaylink-driver-5.6.1/displaylink-installer.sh
patching file 5.6.1/displaylink-driver-5.6.1/displaylink-installer.sh

Installing driver version: 5.6.1

Distribution discovered: No LSB modules are available.
Kali GNU/Linux Rolling

Installing
[ Installing EVDI ]
./displaylink-installer.sh: line 44: /home/deadcow/Downloads/DisplayLink: No such file or directory
[[ Installing EVDI DKMS module ]]
Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/evdi/1.12.0/source -> /usr/src/evdi-1.12.0

Building module:
Cleaning build area...
make -j8 KERNELRELEASE=6.1.0-kali5-amd64 all INCLUDEDIR=/lib/modules/6.1.0-kali5-amd64/build/include KVERSION=6.1.0-kali5-amd64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.0-kali5-amd64 (x86_64)
Consult /var/log/displaylink/evdi_install_make.2023-03-08-1429.log for more information.
Deleting module evdi-1.12.0 completely from the DKMS tree.
ERROR: Failed to install evdi to the kernel tree.
Adding udlfb to blacklist
Adding udl to blacklist

-------------------------------------------------------------------

Performing post install steps

ln: failed to create symbolic link '/opt/displaylink/libstdc++.so.6': No such file or directory

[EVDI Installation Log]:

cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.1.0-kali5-common/scripts/Makefile.build:255: /var/lib/dkms/evdi/1.12.0/build/evdi_fb.o] Error 1
In file included from /var/lib/dkms/evdi/1.12.0/build/evdi_gem.c:24:
/var/lib/dkms/evdi/1.12.0/build/evdi_drm_drv.h:87:32: error: field ‘base’ has incomplete type
   87 |         struct drm_framebuffer base;
      |                                ^~~~
make[2]: *** [/usr/src/linux-headers-6.1.0-kali5-common/scripts/Makefile.build:255: /var/lib/dkms/evdi/1.12.0/build/evdi_gem.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.1.0-kali5-common/Makefile:2030: /var/lib/dkms/evdi/1.12.0/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-kali5-amd64'
make: *** [Makefile:33: all] Error 2

sansquoi, 2023-03-08:
Complete log (evdi_install_make.2023-03-08-1455.log):

DKMS make.log for evdi-1.12.0 for kernel 6.1.0-kali5-amd64 (x86_64)
Wed Mar  8 14:55:07 MST 2023
make KBUILD_VERBOSE=1 M=/var/lib/dkms/evdi/1.12.0/build SUBDIRS=/var/lib/dkms/evdi/1.12.0/build SRCROOT=/var/lib/dkms/evdi/1.12.0/build CONFIG_MODULE_SIG= -C /lib/modules/6.1.0-kali5-amd64/build modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.0-kali5-amd64'
make -f /usr/src/linux-headers-6.1.0-kali5-common/scripts/Makefile.build obj=/var/lib/dkms/evdi/1.12.0/build need-builtin=1 need-modorder=1 
  printf '%s
'   evdi_platform_drv.o evdi_platform_dev.o evdi_sysfs.o evdi_modeset.o evdi_connector.o evdi_encoder.o evdi_drm_drv.o evdi_fb.o evdi_gem.o evdi_painter.o evdi_params.o evdi_cursor.o evdi_debug.o evdi_i2c.o evdi_ioc32.o | awk '!x[$0]++ { print("/var/lib/dkms/evdi/1.12.0/build/"$0) }' > /var/lib/dkms/evdi/1.12.0/build/evdi.mod
   gcc-12 -Wp,-MMD,/var/lib/dkms/evdi/1.12.0/build/.evdi_platform_drv.o.d -nostdinc -I/usr/src/linux-headers-6.1.0-kali5-common/arch/x86/include -I./arch/x86/include/generated -I/usr/src/linux-headers-6.1.0-kali5-common/include -I./include -I/usr/src/linux-headers-6.1.0-kali5-common/arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I/usr/src/linux-headers-6.1.0-kali5-common/include/uapi -I./include/generated/uapi -include /usr/src/linux-headers-6.1.0-kali5-common/include/linux/compiler-version.h -include /usr/src/linux-headers-6.1.0-kali5-common/include/linux/kconfig.h -include /usr/src/linux-headers-6.1.0-kali5-common/include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=/usr/src/linux-headers-6.1.0-kali5-common/= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu11 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -mfunction-return=thunk-extern -fno-jump-tables -mharden-sls=all -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=2048 -fstack-protector-strong -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-dangling-pointer -ftrivial-auto-var-init=zero -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-stringop-truncation -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -Wno-array-bounds -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -g -Iinclude/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_platform_drv"' -DKBUILD_MODNAME='"evdi"' -D__KBUILD_MODNAME=kmod_evdi -c -o /var/lib/dkms/evdi/1.12.0/build/evdi_platform_drv.o /var/lib/dkms/evdi/1.12.0/build/evdi_platform_drv.c   ; ./tools/objtool/objtool --hacks=jump_label --hacks=noinstr --orc --retpoline --rethunk --sls --static-call --uaccess   --module /var/lib/dkms/evdi/1.12.0/build/evdi_platform_drv.o
   gcc-12 -Wp,-MMD,/var/lib/dkms/evdi/1.12.0/build/.evdi_platform_dev.o.d -nostdinc -I/usr/src/linux-headers-6.1.0-kali5-common/arch/x86/include -I./arch/x86/include/generated -I/usr/src/linux-headers-6.1.0-kali5-common/include -I./include -I/usr/src/linux-headers-6.1.0-kali5-common/arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I/usr/src/linux-headers-6.1.0-kali5-common/include/uapi -I./include/generated/uapi -include /usr/src/linux-headers-6.1.0-kali5-common/include/linux/compiler-version.h -include /usr/src/linux-headers-6.1.0-kali5-common/include/linux/kconfig.h -include /usr/src/linux-headers-6.1.0-kali5-common/include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=/usr/src/linux-headers-6.1.0-kali5-common/= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu11 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -mfunction-return=thunk-extern -fno-jump-tables -mharden-sls=all -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=2048 -fstack-protector-strong -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-dangling-pointer -ftrivial-auto-var-init=zero -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-stringop-truncation -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -Wno-array-bounds -Wno-alloc-size-larger [... cut ...]
/var/lib/dkms/evdi/1.12.0/build/evdi_fb.c:350:1: error: control reaches end of non-void function [-Werror=return-type]
  350 | }
      | ^
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.1.0-kali5-common/scripts/Makefile.build:255: /var/lib/dkms/evdi/1.12.0/build/evdi_fb.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.1.0-kali5-common/Makefile:2030: /var/lib/dkms/evdi/1.12.0/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-kali5-amd64'
make: *** [Makefile:33: all] Error 2

Barabazs, 2023-03-09:
I just pushed a minor fix. It's possible that it applies to your situation.

Barabazs, 2023-06-04:
@tapsu01 Please open a new issue.

closed 2023-06-04 by Barabazs