AdnanHodzic/displaylink-debian #802: failure on Debian 11

loveshack, 2023-03-09:
On Debian 11, running ./displaylink-debian.sh first fails with 
```
Performing post install steps

ln: failed to create symbolic link '/opt/displaylink/libstdc++.so.6': No such file or directory
```
After creating /opt/displaylink it then stops with 
```
Performing post install steps

sed: can't read /lib/systemd/system/displaylink-driver.service: No such file or directory
```

Sorry, I don't need it badly enough to look at fixing that.

Here's the debug info:
```
--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-21-amd64

---------------- DisplayLink info ----------------

Driver version: 1.9.0+dfsg
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
Subsystem: GT2
VGA: Intel Corporation TigerLake GT2 [Iris Xe Graphics] (rev 01)
VGA (3D): 
X11 version: 1.20.11-1+deb11u5

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x49 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 4 outputs: 6 associated providers: 0 name:modesetting
```

DaniJohni, 2023-03-24:
I have the same issue when trying to install

loveshack, 2023-05-23:
The current (2023-04-21) version works.

closed 2023-05-23 by loveshack