AdnanHodzic/displaylink-debian #803: Fails on Ubuntu 22.04

lpegoraro, 2023-03-20:
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: jammy
Kernel: 6.2.0-76060200-generic

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: [1558:867c]
[GeForce
VGA: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
NVIDIA Corporation GA106M [GeForce RTX 3060 Mobile / Max-Q] (rev a1)
VGA (3D): 
X11 version: 21.1.3-2ubuntu2.7

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 5 associated providers: 1 name:modesetting
Provider 1: id: 0x201 cap: 0x2, Sink Output crtcs: 4 outputs: 4 associated providers: 1 name:NVIDIA-G0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

junocomp, 2023-03-25:
I am having the same issue

Barabazs, 2023-06-04:
What is the content of `/var/log/displaylink/evdi_install_make.2023-03-20-1802.log`?

cojocar, 2023-06-24:
Fails on Debian Bookworm too, with the same message. The kernel module fails to build because `drm.h` is missing. I have installed `libdrm-dev` and it still fails. In adition, it fails too by using the ["official" method](https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu). Probably this is a evdi bug, anyway log follows.
```
DKMS make.log for evdi-1.13.1 for kernel 6.1.0-9-amd64 (x86_64)
Sat Jun 24 09:09:31 PDT 2023
make KBUILD_VERBOSE=1 M=/var/lib/dkms/evdi/1.13.1/build SUBDIRS=/var/lib/dkms/evdi/1.13.1/build SRCROOT=/var/lib/dkms/evdi/1.13.1/build CONFIG_MODULE_SIG= -C /lib/modules/6.1.0-9-amd64/build modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.0-9-amd64'
make -f /usr/src/linux-headers-6.1.0-9-common/scripts/Makefile.build obj=/var/lib/dkms/evdi/1.13.1/build need-builtin=1 need-modorder=1 
  printf '%s
'   evdi_platform_drv.o evdi_platform_dev.o evdi_sysfs.o evdi_modeset.o evdi_connector.o evdi_encoder.o evdi_drm_drv.o evdi_fb.o evdi_gem.o evdi_painter.o evdi_params.o evdi_cursor.o evdi_debug.o evdi_i2c.o evdi_ioc32.o | awk '!x[$0]++ { print("/var/lib/dkms/evdi/1.13.1/build/"$0) }' > /var/lib/dkms/evdi/1.13.1/build/evdi.mod
   gcc-12 -Wp,-MMD,/var/lib/dkms/evdi/1.13.1/build/.evdi_platform_drv.o.d -nostdinc -I/usr/src/linux-headers-6.1.0-9-common/arch/x86/include -I./arch/x86/include/generated -I/usr/src/linux-headers-6.1.0-9-common/include -I./include -I/usr/src/linux-headers-6.1.0-9-common/arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I/usr/src/linux-headers-6.1.0-9-common/include/uapi -I./include/generated/uapi -include /usr/src/linux-headers-6.1.0-9-common/include/linux/compiler-version.h -include /usr/src/linux-headers-6.1.0-9-common/include/linux/kconfig.h -include /usr/src/linux-headers-6.1.0-9-common/include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=/usr/src/linux-headers-6.1.0-9-common/= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu11 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -mfunction-return=thunk-extern -fno-jump-tables -mharden-sls=all -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=2048 -fstack-protector-strong -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-dangling-pointer -ftrivial-auto-var-init=zero -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-stringop-truncation -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -Wno-array-bounds -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -g -Iinclude/uapi/drm -Iinclude/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_platform_drv"' -DKBUILD_MODNAME='"evdi"' -D__KBUILD_MODNAME=kmod_evdi -c -o /var/lib/dkms/evdi/1.13.1/build/evdi_platform_drv.o /var/lib/dkms/evdi/1.13.1/build/evdi_platform_drv.c   ; ./tools/objtool/objtool --hacks=jump_label --hacks=noinstr --orc --retpoline --rethunk --sls --static-call --uaccess   --module /var/lib/dkms/evdi/1.13.1/build/evdi_platform_drv.o
   gcc-12 -Wp,-MMD,/var/lib/dkms/evdi/1.13.1/build/.evdi_platform_dev.o.d -nostdinc -I/usr/src/linux-headers-6.1.0-9-common/arch/x86/include -I./arch/x86/include/generated -I/usr/src/linux-headers-6.1.0-9-common/include -I./include -I/usr/src/linux-headers-6.1.0-9-common/arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I/usr/src/linux-headers-6.1.0-9-common/include/uapi -I./include/generated/uapi -include /usr/src/linux-headers-6.1.0-9-common/include/linux/compiler-version.h -include /usr/src/linux-headers-6.1.0-9-common/include/linux/kconfig.h -include /usr/src/linux-headers-6.1.0-9-common/include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=/usr/src/linux-headers-6.1.0-9-common/= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu11 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -mfunction-return=thunk-extern -fno-jump-tables -mharden-sls=all -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=2048 -fstack-protector-strong -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-dangling-pointer -ftrivial-auto-var-init=zero -fno-stack-clash-protection -pg -mrecord-mc [... cut ...] not-aligned -g -Iinclude/uapi/drm -Iinclude/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_gem"' -DKBUILD_MODNAME='"evdi"' -D__KBUILD_MODNAME=kmod_evdi /var/lib/dkms/evdi/1.13.1/build/evdi_gem.c | scripts/genksyms/genksyms   -r /dev/null >> /var/lib/dkms/evdi/1.13.1/build/.evdi_gem.o.cmd; fi
make[1]: *** [/usr/src/linux-headers-6.1.0-9-common/Makefile:2037: /var/lib/dkms/evdi/1.13.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-9-amd64'
make: *** [Makefile:38: all] Error 2
```

Barabazs, 2024-05-03:
Should be resolved with latest release.

closed 2024-05-03 by Barabazs