AdnanHodzic/displaylink-debian #805: Debian 11 Bullseye ln: failed to create symbolic link '/opt/displaylink/libstdc++.so.6': No such file or directory

dev2devportal, 2023-03-31:
Can't complete installation:
`Performing post install steps
ln: failed to create symbolic link '/opt/displaylink/libstdc++.so.6': No such file or directory`

`git clone https://github.com/AdnanHodzic/displaylink-debian
Cloning into 'displaylink-debian'...
remote: Enumerating objects: 1180, done.
remote: Counting objects: 100% (247/247), done.
remote: Compressing objects: 100% (131/131), done.
remote: Total 1180 (delta 142), reused 202 (delta 114), pack-reused 933
Receiving objects: 100% (1180/1180), 290.11 KiB | 3.67 MiB/s, done.
Resolving deltas: 100% (548/548), done.
root@strx:/home/bofh/Downloads/DisplayLink-git# ls
displaylink-debian  displaylink-debian-fail  displaylink-debian-fail2
root@strx:/home/bofh/Downloads/DisplayLink-git# cd displaylink-debian
root@strx:/home/bofh/Downloads/DisplayLink-git/displaylink-debian# ls
displaylink-debian.sh  displaylink.sh  LICENSE  post-install-guide.md  README.md  resources  self-signing-modules.md

# ./displaylink-debian.sh

--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: I

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-6.0.0-0.deb11.6-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
libelf-dev is installed
git is installed
pciutils is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at:
https://www.synaptics.com/node/4781?filetype=exe
Do you accept?: [Y/n]: Y

Downloading DisplayLink Ubuntu driver:

--2023-03-31 15:05:07--  https://www.synaptics.com/sites/default/files/exe_files/2022-08/DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu5.6.1-EXE.zip
Resolving www.synaptics.com (www.synaptics.com)... 54.245.106.105
Connecting to www.synaptics.com (www.synaptics.com)|54.245.106.105|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 16905857 (16M) [application/zip]
Saving to: ‘DisplayLink_Ubuntu_5.6.1.zip’

DisplayLink_Ubuntu_5.6.1.zip  100%[=================================================>]  16.12M  9.81MB/s    in 1.6s

2023-03-31 15:05:09 (9.81 MB/s) - ‘DisplayLink_Ubuntu_5.6.1.zip’ saved [16905857/16905857]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.6.1.zip
  inflating: 5.6.1/displaylink-driver-5.6.1-59.184.run
  inflating: 5.6.1/DisplayLink USB Graphics Software for Ubuntu 5.6.1-59.185-Release Notes.txt
  inflating: 5.6.1/__MACOSX/._DisplayLink USB Graphics Software for Ubuntu 5.6.1-59.185-Release Notes.txt
Creating directory displaylink-driver-5.6.1-59.184
Verifying archive integrity...  100%   MD5 checksums are OK. All good.
Uncompressing DisplayLink Linux Driver 5.6.1-59.184  100%
patching file 5.6.1/displaylink-driver-5.6.1/displaylink-installer.sh
patching file 5.6.1/displaylink-driver-5.6.1/displaylink-installer.sh

Installing driver version: 5.6.1

Distribution discovered: Debian GNU/Linux 11 (bullseye)
WARNING: Version 1.12.0 of EVDI kernel module is already running.
Please reboot before attempting to re-install DisplayLink Linux Software.
Installation terminated.
Adding udlfb to blacklist
Adding udl to blacklist

-------------------------------------------------------------------

Performing post install steps

ln: failed to create symbolic link '/opt/displaylink/libstdc++.so.6': No such file or directory`

# uname -a
Linux strx 6.0.0-0.deb11.6-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.0.12-1~bpo11+1 (2022-12-19) x86_64 GNU/Linux

# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

decadenza, 2023-04-04:
Identical issue here.

dev2devportal, 2023-04-24:
Any chance of this issue getting some attention?

ZiTAL, 2023-06-14:
`
su
apt-get install libstdc++6
ln -s /usr/lib/x86_64-linux-gnu/libstdc++.so.6 /opt/displaylink/libstdc++.so.6
`

and try again

fgkolf, 2023-07-13:
I bypassed this by running `mkdir -p /opt/displaylink` before installation

Barabazs, 2024-05-03:
Closing this issue in favor of new meta issue where I'll group all related issues.

closed 2024-05-03 by Barabazs