AdnanHodzic/displaylink-debian #806: Docking station problem?

asarnacki, 2023-04-04:
Hi, I have an i-tec USB 3.0 Dual Docking Station HDMI DVI docking station, unfortunately, after installing the drivers from the manufacturer, neither the monitors nor the internet via cable work, I tried to install the displaylink drivers from the community, unfortunately the same thing, I tried to change the configuration described in this thread https://github.com/AdnanHodzic/displaylink-debian/issues/228#issuecomment-467889348 and this https://github.com/AdnanHodzic/displaylink-debian/issues/236#issuecomment-471213411 , it didn't help, I have no idea, can I do anything else?

```
--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.15.0-69-generic

---------------- DisplayLink info ----------------

Driver version: 1.12.0
1.7.0
DisplayLink service status: up and running
EVDI service version: 1.12.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 
VGA: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] (rev 01)
VGA (3D): 
X11 version: 1.20.13-1ubuntu1~20.04.8
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x45 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 2 associated providers: 4 name:modesetting
Provider 1: id: 0x13a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x118 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xf6 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xd4 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------
```

Barabazs, 2023-06-04:
The debug information shows that displaylink & evdi are successfully installed and the monitors/providers are also associated as they should. There must be another issue with the docking station itself. Please raise an issue with the reseller.

closed 2023-06-04 by Barabazs

mikelgutizubi, 2023-08-10:
Very similar issue in #858