AdnanHodzic/displaylink-debian #811: Kernel 6.1.0-7-amd64 (Debian 6.1.20-1) errors: field ‘base’ has incomplete type; expression in static assertion is not an integer; invalid use of undefined type ‘struct drm_framebuffer’

marcelolaia, 2023-04-12:
```
error: field ‘base’ has incomplete type
87 |         struct drm_framebuffer base;

```
(...)

```
error: expression in static assertion is not an integer 
298 | #define __same_type(a, b) __builtin_types_compatible_p(typeof(a), typeof(b))

```
(...)

```
error: invalid use of undefined type ‘struct drm_framebuffer’
271 |                         0, 0, fb->width, fb->height

```
(...)

`ln: failed to create symbolic link '/opt/displaylink/libstdc++.so.6': No such file or directory`

Here is the complete log

[evdi_install_make.2023-04-12-1647.log](https://github.com/AdnanHodzic/displaylink-debian/files/11215448/evdi_install_make.2023-04-12-1647.log)

brenard, 2023-04-15:
Same here and I noticed that the `evdi-dkms` debian package with the same version of the module (1.12.0) could be compiled without problem. Their [patch](https://sources.debian.org/patches/evdi/1.12.0%2Bdfsg-0.3/0001-fix-6.0-ftbfs.patch/) for v6 kernel compatibility is smaller that one included with displaylink-debian (`resources/evdi.patch`), so I tried to replace it and it fix the problem.

My version of the `resources/evdi.patch` :

```
--- module/evdi_drm_drv.h.orig	2023-04-15 18:01:08.183472491 +0200
+++ module/evdi_drm_drv.h	2023-04-15 18:01:37.016119676 +0200
@@ -26,7 +26,7 @@
 #include <drm/drmP.h>
 #endif
 #if KERNEL_VERSION(5, 15, 0) <= LINUX_VERSION_CODE
-#include <drm/drm_legacy.h>
+#include <drm/drm_framebuffer.h>
 #else
 #include <drm/drm_irq.h>
 #endif
```

marcelolaia, 2023-04-17:
Didn't work here.

Here is my `evdi_drm_drv.h`

```
#if KERNEL_VERSION(5, 5, 0) <= LINUX_VERSION_CODE || defined(EL8)
#include <drm/drm_drv.h>
#include <drm/drm_fourcc.h>
#include <drm/drm_ioctl.h>
#include <drm/drm_vblank.h>
#else
#include <drm/drmP.h>
#endif
#if KERNEL_VERSION(5, 15, 0) <= LINUX_VERSION_CODE
#include <drm/drm_framebuffer.h>
#else
#include <drm/drm_irq.h>
#endif
#include <drm/drm_crtc.h>
#include <drm/drm_crtc_helper.h>
#include <drm/drm_rect.h>
#include <drm/drm_gem.h>
#include <drm/drm_framebuffer.h>
#if KERNEL_VERSION(5, 4, 0) <= LINUX_VERSION_CODE || defined(EL8)
```

closed 2023-04-20 by Barabazs

marcelolaia, 2023-04-21:
Problem presist! Here is the `/var/lib/dkms/evdi/1.13.1/build/make.log`:

```
DKMS make.log for evdi-1.13.1 for kernel 6.1.0-7-amd64 (x86_64)
Fri Apr 21 13:56:00 -03 2023
make KBUILD_VERBOSE=1 M=/var/lib/dkms/evdi/1.13.1/build SUBDIRS=/var/lib/dkms/evdi/1.13.1/build SRCROOT=/var/lib/dkms/evdi/1.13.1/build CONFIG_MODULE_SIG= -C /lib/modules/6.1.0-7-amd64/build modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.0-7-amd64'
make -f /usr/src/linux-headers-6.1.0-7-common/scripts/Makefile.build obj=/var/lib/dkms/evdi/1.13.1/build need-builtin=1 need-modorder=1 
  printf '%s
'   evdi_platform_drv.o evdi_platform_dev.o evdi_sysfs.o evdi_modeset.o evdi_connector.o evdi_encoder.o evdi_drm_drv.o evdi_fb.o evdi_gem.o evdi_painter.o evdi_params.o evdi_cursor.o evdi_debug.o evdi_i2c.o evdi_ioc32.o | awk '!x[$0]++ { print("/var/lib/dkms/evdi/1.13.1/build/"$0) }' > /var/lib/dkms/evdi/1.13.1/build/evdi.mod
   gcc-12 -Wp,-MMD,/var/lib/dkms/evdi/1.13.1/build/.evdi_platform_drv.o.d -nostdinc -I/usr/src/linux-headers-6.1.0-7-common/arch/x86/include -I./arch/x86/include/generated -I/usr/src/linux-headers-6.1.0-7-common/include -I./include -I/usr/src/linux-headers-6.1.0-7-common/arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I/usr/src/linux-headers-6.1.0-7-common/include/uapi -I./include/generated/uapi -include /usr/src/linux-headers-6.1.0-7-common/include/linux/compiler-version.h -include /usr/src/linux-headers-6.1.0-7-common/include/linux/kconfig.h -include /usr/src/linux-headers-6.1.0-7-common/include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=/usr/src/linux-headers-6.1.0-7-common/= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu11 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -mfunction-return=thunk-extern -fno-jump-tables -mharden-sls=all -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=2048 -fstack-protector-strong -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-dangling-pointer -ftrivial-auto-var-init=zero -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-stringop-truncation -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -Wno-array-bounds -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -g -Iinclude/uapi/drm -Iinclude/drm  -DMODULE  -DKBUILD_BASENAME='"evdi_platform_drv"' -DKBUILD_MODNAME='"evdi"' -D__KBUILD_MODNAME=kmod_evdi -c -o /var/lib/dkms/evdi/1.13.1/build/evdi_platform_drv.o /var/lib/dkms/evdi/1.13.1/build/evdi_platform_drv.c   ; ./tools/objtool/objtool --hacks=jump_label --hacks=noinstr --orc --retpoline --rethunk --sls --static-call --uaccess   --module /var/lib/dkms/evdi/1.13.1/build/evdi_platform_drv.o
   gcc-12 -Wp,-MMD,/var/lib/dkms/evdi/1.13.1/build/.evdi_platform_dev.o.d -nostdinc -I/usr/src/linux-headers-6.1.0-7-common/arch/x86/include -I./arch/x86/include/generated -I/usr/src/linux-headers-6.1.0-7-common/include -I./include -I/usr/src/linux-headers-6.1.0-7-common/arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I/usr/src/linux-headers-6.1.0-7-common/include/uapi -I./include/generated/uapi -include /usr/src/linux-headers-6.1.0-7-common/include/linux/compiler-version.h -include /usr/src/linux-headers-6.1.0-7-common/include/linux/kconfig.h -include /usr/src/linux-headers-6.1.0-7-common/include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=/usr/src/linux-headers-6.1.0-7-common/= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu11 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -mfunction-return=thunk-extern -fno-jump-tables -mharden-sls=all -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -Wno-address-of-packed-member -O2 -fno-allow-store-data-races -Wframe-larger-than=2048 -fstack-protector-strong -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-dangling-pointer -ftrivial-auto-var-init=zero -fno-stack-clash-protection -pg -mrecord-mcount -mfentry -DCC_USING_FENTRY -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-stringop-truncation -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -Wno-array-bounds -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -fno-st [... cut ...]
+  # printf "\n\n%s\n%s\n%s\n\n\n" "Reboot required" "DisplayLink driver installed successfully." \
+  #       "Please reboot your computer to ensure the proper functioning of the software." \
+  #     && prompt_yes_no "Do you want to reboot now?" \
+  #       && reboot
 
-  "$NOREBOOT" && exit 0
+  # "$NOREBOOT" && exit 0
 
-  xorg_running || exit 0
-  prompt_yes_no "Xorg is running. Do you want to reboot now?" && reboot
+  # xorg_running || exit 0
+  # prompt_yes_no "Xorg is running. Do you want to reboot now?" && reboot
   exit 0
 }
```