AdnanHodzic/displaylink-debian #815: I cant turn on DVI devices

richardd82, 2023-04-17:
Hello, tahn you for you time. I have installed your desplaylink-debian driver and i can see that i have new devices but i cant turn on them. My log after --debug option is this:

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: jammy
Kernel: 5.19.0-38-generic

---------------- DisplayLink info ----------------

Driver version: 1.12.0
DisplayLink service status: up and running
EVDI service version: 1.12.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 530
VGA: Intel Corporation HD Graphics 530 (rev 06)
VGA (3D): 
X11 version: 21.1.4-2ubuntu1.7~22.04.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 2 outputs: 6 associated providers: 4 name:modesetting
Provider 1: id: 0x101 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xe0 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xbf cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x9e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

richardd82@richardd82-Inspiron-7348:~/displaylink-debian$ xrandr --listproviders
Providers: number : 5
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 2 outputs: 6 associated providers: 4 name:modesetting
Provider 1: id: 0x101 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xe0 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xbf cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x9e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
richardd82@richardd82-Inspiron-7348:~/displaylink-debian$ xrandr --setprovideroutputsource 1 0
xrandr --setprovideroutputsource 2 0
xrandr --setprovideroutputsource 3 0
xrandr --setprovideroutputsource 4 0
richardd82@richardd82-Inspiron-7348:~/displaylink-debian$ xrandr
Screen 0: minimum 320 x 200, current 3280 x 1080, maximum 16384 x 16384
HDMI-1 connected 1360x768+0+0 (normal left inverted right x axis y axis) 698mm x 392mm
   1920x1080     60.00 +  59.94    30.00    24.00    29.97    23.98  
   1920x1080i    60.00    59.94  
   1680x1050     59.88  
   1280x1024     75.02    60.02  
   1440x900      59.90  
   1280x960      60.00  
   1360x768      60.02* 
   1280x800      59.91  
   1152x864      75.00  
   1280x720      60.00    60.00    30.00    59.94    29.97    24.00    23.98  
   1024x768      75.03    70.07    60.00  
   800x600       72.19    75.00    60.32  
   720x480       60.00    59.94  
   720x480i      60.00    59.94  
   640x480       75.00    72.81    60.00    59.94  
   720x400       70.08  
DP-1 connected primary 1920x1080+1360+0 (normal left inverted right x axis y axis) 476mm x 268mm
   1920x1080     60.00*+
   1600x900      60.00  
   1280x1024     75.02    60.02  
   1152x864      75.00  
   1024x768      75.03    60.00  
   800x600       75.00    60.32  
   640x480       75.00    59.94  
   720x400       70.08  
HDMI-2 disconnected (normal left inverted right x axis y axis)
DP-2 disconnected (normal left inverted right x axis y axis)
HDMI-3 disconnected (normal left inverted right x axis y axis)
DP-3 disconnected (normal left inverted right x axis y axis)
DVI-I-4-4 disconnected (normal left inverted right x axis y axis)
DVI-I-3-3 disconnected (normal left inverted right x axis y axis)
DVI-I-2-2 disconnected (normal left inverted right x axis y axis)
DVI-I-1-1 disconnected (normal left inverted right x axis y axis)

I followed the post install instructions and the documentation you provide from the official xrandr site but I still can't see my other 2 monitors. Thank you very much for your support

kaiquepy, 2024-02-06:
I have the same issue. Any updates?

richardd82, 2024-02-07:
No, finally i bougth a rtx 4060 to connect my four monitors.

closed 2024-02-07 by richardd82