AdnanHodzic/displaylink-debian #816: Install script fails

plipK, 2023-04-20:
I failed to install the driver with the ./displaylink-debian.sh  script (commit 7d7d1804066c2522d6a59e26696dfddaceb2fac4):
```

Archive:  DisplayLink_Ubuntu_5.7.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of DisplayLink_Ubuntu_5.7.zip or
        DisplayLink_Ubuntu_5.7.zip.zip, and cannot find DisplayLink_Ubuntu_5.7.zip.ZIP, period.
```
The .zip file is not valid, it contains html code.
It seems like the download links are not created correctly. I added the following statements:

```
echo "versions: $versions"
echo "dlurl: $dlurl"
echo "driver_url: $driver_url"
```

And this is the output:

```
versions: 5.75.6.1
dlurl: https://www.synaptics.com/
driver_url: https://www.synaptics.com/
```

Debug output:

```
sudo ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-21-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: nvidia
Subsystem: T2000
VGA: NVIDIA Corporation TU117GLM [Quadro T2000 Mobile / Max-Q] (rev a1)
VGA (3D): 
X11 version: 1.20.11-1+deb11u5
X11 configs: /etc/X11/xorg.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: Datei oder Verzeichnis nicht gefunden
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x218 cap: 0x1, Source Output crtcs: 4 outputs: 8 associated providers: 0 name:NVIDIA-0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

losocio, 2023-04-20:
Wow, what a coincidence someone is having the same issue and posting about it just 1 hour ago. I'm trying to get the drivers installed on my machine running Linux Mint 21.2 Vera and I run into the same problem:
`Archive:  DisplayLink_Ubuntu_5.7.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of DisplayLink_Ubuntu_5.7.zip or
        DisplayLink_Ubuntu_5.7.zip.zip, and cannot find DisplayLink_Ubuntu_5.7.zip.ZIP, period.
`
My debug output is:
`
sudo ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Linuxmint
Release: vera
Kernel: 5.15.0-70-generic

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: 1.13.1

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: GT2
[GeForce
VGA: Intel Corporation CoffeeLake-H GT2 [UHD Graphics 630]
NVIDIA Corporation GP107M [GeForce GTX 1050 Mobile] (rev a1)
VGA (3D): 
X11 version: 21.1.4-2ubuntu1.7~22.04.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x44 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 3 associated providers: 5 name:modesetting
Provider 1: id: 0x27e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x25d cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x23c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x21b cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1f3 cap: 0x2, Sink Output crtcs: 4 outputs: 3 associated providers: 1 name:NVIDIA-G0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
`
I hope we both get to resolve this issue...

Barabazs, 2023-04-20:
I updated the code to download the correct version.
Please try again with latest changes.

closed 2023-04-20 by Barabazs

plipK, 2023-04-20:
It still doesn't work for me. I get the same error message.

Barabazs, 2023-04-21:
I made new changes and it successfully installed on my test machines.
Sorry for the trouble.

 Please try again with latest changes.

plipK, 2023-04-21:
It works. Thanks!