AdnanHodzic/displaylink-debian #819: DisplayLink_Ubuntu_5.7.zip download problems

marcelolaia, 2023-04-20:
`sudo ./displaylink-debian.sh`

```
Preparing for install

Archive:  DisplayLink_Ubuntu_5.7.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of DisplayLink_Ubuntu_5.7.zip or
        DisplayLink_Ubuntu_5.7.zip.zip, and cannot find DisplayLink_Ubuntu_5.7.zip.ZIP, period.

```

Is there a possibility to download the driver manually and point the script to install it instead re-download?

jotapikie, 2023-04-20:
I am having the same issue

standevenjw, 2023-04-20:
There are a few issues with the update to the driver package.

I had to change the dlurl variable by adding a grep for href="/node.

        dlurl="https://www.synaptics.com/$(wget -q -O - https://www.synaptics.com/products/displaylink-graphics/downloads/ubuntu | grep 'class="download-link"' | grep 'href="/node' | head -n 1 | perl -pe '($_)=/<a href="\/([^"]+)"[^>]+class="download-link"/')"

The inflated file name has also changed.

    chmod +x $driver_dir/displaylink-driver-${version}.[0-9]*.[0-9]*.run
    ./$driver_dir/displaylink-driver-${version}.[0-9]*.[0-9]*.run --keep --noexec
    mv displaylink-driver-${version}.*[0-9]*.[0-9]*/ $driver_dir/displaylink-driver-${version}

Note that between the ${version} and [0-9] it is now a period instead of a dash.

And finally the patch is no longer valid so I commented it out.

    # Patch displaylink-installer.sh to prevent reboot before our script is done.
    #patchName="displaylink-installer.patch"
    #finalPatchPath="$resourcesDir$patchName"
    #patch -Np0 $driver_dir/displaylink-driver-${version}/displaylink-installer.sh <$finalPatchPath

Barabazs, 2023-04-21:
I made new changes and it successfully installed on my test machines.
Sorry for the trouble.

 Please try again with latest changes.

closed 2023-04-21 by Barabazs