AdnanHodzic/displaylink-debian #821: Error make

Nicolas-64, 2023-04-21:
Hello,

I have an error during the installation on Debian 11: 

make -j8 KERNELRELEASE=5.10.0-21-amd64 all INCLUDEDIR=/lib/modules/5.10.0-21-amd64/build/include KVERSION=5.10.0-21-amd64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.0-21-amd64 (x86_64)

In /var/lib/dkms/evdi/1.13.1/build/make.log the following message appears:

echo >&2 " ERROR: Kernel configuration is invalid."; \
echo >&2 " include/generated/autoconf.h or include/config/auto.conf are missing.
echo >&2 " Run 'make oldconfig && make prepare' on kernel src to fix it."; \

Could you please tell me where this error could be coming from.

Thank you very much,
Nicolas

Nicolas-64, 2023-04-23:
Hello,

After having done the following:
- Extraction of the kernel sources
- Creation of the .config file
- Run "make oldconfig".
- Execution "make prepare".

The installation seems to have worked well.

I can't test the screens yet, I'm not in the office on Sunday ;-) but I'll test it tomorrow morning.

Thank you,
Nicolas

Nicolas-64, 2023-04-24:
Hello,
It works, my screens are well detected and work.

tshelley, 2023-04-25:
@Nicolas-64

Hi there, I am getting the same error.

Do you mind expounding on the first bullet?

Should all of these steps/commands be in the /var/lib/dkms/evdi/1.13.1/build directory?

Thank you,

Tom

Nicolas-64, 2023-04-25:
Hello,

As indicated in the log, you should have done "make oldconfig && make prepare".

But before I had to :
- get the kernel sources 
  cd /usr/src/linux-headers-5.10.0-21-amd64 ( depending on the kernel version )
  tar -xaf /usr/src/linux-source-5.10.tar.xz

- get the last configuration file :
cp /boot/config-5.10.0-8-amd64 /usr/src/linux-headers-5.10.0-21-amd64/.config

Then I could do in the /usr/src/linux-headers-5.10.0-21-amd64 folder :
  make oldconfig
  make prepare

And finally I uninstalled and reinstalled the DisplayLink drivers

I hope this will help you.

Nicolas

pitch-d, 2023-04-28:
Thank you @Nicolas-64 
I'm having the same issue but I can't make your instruction work
Something is wrong with debian kernel packages to begin with. This should work out of the box imo

The first point, if I `tar xf /usr/src/linux-source-5.10.tar.xz` when standing in `/usr/src/linux-headers-5.10.0-21-amd64` i will get a folder named `linux-source-5.10` in that folder and make will not find those files
how did you solve that?
when I instead run `tar xf /usr/src/linux-source-5.10.tar.xz --strip-components=1` it will replace the files in `/usr/src/linux-headers-5.10.0-21-amd64` maybe not perfect but that might get me there I thought, but no
make oldconfig works
make prepare says:
`  HOSTCC  scripts/sign-file
scripts/sign-file.c:25:10: fatal error: openssl/opensslv.h: No such file or directory
   25 | #include <openssl/opensslv.h>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.host:95: scripts/sign-file] Error 1
make: *** [Makefile:1203: scripts] Error 2`

I can't find out how this is meant to work and as I understand displaylink can't be installed on debian 11 atm without reconfigure debian kernel packages manually

Any suggestions how I can get my displaylink working on debian 11 are welcome

tshelley, 2023-05-12:
@Nicolas-64 Thanks for the guidance. I haven't had a chance until now to try this.

vallezwh, 2023-05-12:
cd /usr/src/linux-headers-5.10.0-22-amd64

tar -xaf /usr/src/linux-source-5.10.tar.xz

sudo cp /boot/config-5.10.0-22-amd64 .config

sudo make oldconfig
sh: 0: cannot open /usr/src/linux-headers-5.10.0-22-common/scripts/mkmakefile: No such file
make[1]: *** [/usr/src/linux-headers-5.10.0-22-common/Makefile:571: outputmakefile] Error 2
make: *** [/usr/src/linux-headers-5.10.0-22-common/Makefile:192: __sub-make] Error 2

don't you mean

cd /usr/src/linux-headers-5.10.0-22-amd64/linux-source-5.10

sudo make oldconfig ; sudo make prepare

that executes, but still get the same error with the installer

vallezwh, 2023-05-12:
source had to be deployed in headers directory for it to work for me

stfpz, 2023-05-16:
@vallezwh : 
> source had to be deployed in headers directory for it to work for me

Could you please elaborate a little bit on this? What do you mean by "deployed in headers directory"?

Trying to use @Nicolas-64 's workaround (Debian/bullseye), I got sth like:

```
/usr/src/linux-headers-5.10.0-23-amd64
├── arch
├── include
├── linux-source-5.10
├── Makefile
├── Module.symvers
├── scripts -> ../../lib/linux-kbuild-5.10/scripts
├── source -> /usr/src/linux-headers-5.10.0-23-common
└── tools -> ../../lib/linux-kbuild-5.10/tools
```

After cd-ing into linux-source-5.10 (and installing libssl-dev to get rid of the error that @pitch-d got), the "make oldconfig && make prepare" combo worked, but unfortunately not the `evdi` module compilation during execution of `displaylink-debian.sh` (still fails requesting `make oldconfig` + `prepare`...).

vallezwh, 2023-05-22:
uninstall displaylink and any remnants of evdi
reboot

cd /usr/src/linux-headers-5.10.0-23-amd64
sudo tar -xaf /usr/src/linux-source-5.10.tar.xz --strip-components=1
make oldconfig && make prepare

install displaylink

iamdey, 2023-05-30:
Great, this is probably the most advanced workaround for several issues currently open on this repo for the error: 
> EVDI service version: /sys/devices/evdi/version not found

When installing displaylink in the first place, this error is shown but not blocking:

>Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.10.0-23-amd64 all INCLUDEDIR=/lib/modules/5.10.0-23-amd64/build/include KVERSION=5.10.0-23-amd64 DKMS_BUILD=1...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.0-23-amd64 (x86_64)
Consult /var/lib/dkms/evdi/1.13.1/build/make.log for more information.

I now have evdi correctly installed thanks to this workaround. I'll just re-format the previous comments here for newbies like me:

---

**Prerequisite**: no displaylink must not be installed

```bash
# make sure libssl-dev is installed, which will required in the prepare command
sudo apt install libssl-dev
cd /usr/src/linux-headers-5.10.0-23-amd64
sudo tar -xaf /usr/src/linux-source-5.10.tar.xz --strip-components=1
sudo make oldconfig
sudo make prepare
```

Then install displaylink as usual:

```bash
sudo ./displaylink-debian.sh
```

losocio, 2023-06-01:
Hello, I'm the author of issue #825. 
After reading all the comments in this issue, I tried to do what @iamdey suggested but didn't work. The install worked, but after reboot I had "[FAILED] Failed to start DisplayLink Driver Service" and the OS wouldn't start. So I restored the system to the previous snapshot. I suspect the it has something to do with Linux Mint not running kernel version 5.10.0, but instead 5.15.0. (I gather that many of you are using Debian and not Mint). Any help would be appreciated, I'm a newbie and don't know what I'm doing :)

iamdey, 2023-06-02:
Well my comment probably needs to be adapted to your kernel version, feel free to suggest improvements.

p91paul, 2023-06-03:
@Barabazs you seem to be the maintainer of this project. What is presented here looks like a workaround, can we expect some "official" fix?

Barabazs, 2023-06-04:
Thank you all for the very helpful comments and fixes.

I have documented the workaround (and tried to generalise it) in the docs: https://github.com/AdnanHodzic/displaylink-debian/blob/master/docs/common-issues.md#make-error-bad-return-status-for-module-build-on-kernel

closed 2023-06-04 by Barabazs