AdnanHodzic/displaylink-debian #823: Hybrid Graphics (AMD Renior + NVIDIA)

SZUPEK, 2023-04-22:
I am using non-proprietary drivers but the GPU is tied the the dGPU (which I cant afford the battery drain on). I have a display link adapter hooked up via USB-A but like all things NVIDIA, it won't work.  I want to use the Integrated AMD graphics ONLY (and still have the dGPU needed for a virtual machine, if required). 

This is a business environment. The dGPU is only there incase it's needed.

I've tried NVDA proprietary, and then laptop screen doesn't work. I need both an external montior AND the iGPU to work (again iGPU is AMD Renoir). Laptop is an Asus Studiobook 16 (H5600QM).

LOG below
-----
Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.10.0-21-amd64

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: amdgpu
Subsystem: Device
Device
VGA: NVIDIA Corporation GA106M [GeForce RTX 3060 Mobile / Max-Q] (rev a1)
Advanced Micro Devices, Inc. [AMD/ATI] Cezanne (rev c4)
VGA (3D): 
X11 version: 1.20.11-1+deb11u6
X11 configs: /etc/X11/xorg.conf.d/00-keyboard.conf /etc/X11/xorg.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 0

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

iamdey, 2023-05-30:
This is probably due to an error when building evdi, see #821 
I think this should be closed as duplicate (sorry for the spam, but it seems to be duplicated several times)

Barabazs, 2023-06-04:
@SZUPEK I see that evdi was not installed successfully. Can you repeat the install procedure and copy the terminal output here?

closed 2024-05-03 by Barabazs