AdnanHodzic/displaylink-debian #824: PureOS on Librem5

methuselah-0, 2023-04-24:
I recently tried the 5.7.2 version on PureOS on a librem 5 phone where installation goes fine, but after reboot and manually attempting to start the service with "systemctl start displaylink-driver.service" it says that the evdi driver is not found - it fails at the modprobe evdi step. The evdi.ko file is however present in the /lib/modules/... directory. Any ideas?

Barabazs, 2023-06-04:
Please post the output of `sudo ./displaylink-debian.sh --debug`. You might also want to revert all your changes and try to install again, but this time copy and paste the terminal output here please.

janvlug, 2023-06-04:
See also: https://github.com/AdnanHodzic/displaylink-debian/issues/778

closed 2024-05-03 by Barabazs