AdnanHodzic/displaylink-debian #825: Install is successful after running script. But GUI won't load on reboot.

losocio, 2023-04-26:
Hi,

I've looked around past summited issues and found one similar to my problem. But it seems to have been abandoned by OP.
The issue: https://github.com/AdnanHodzic/displaylink-debian/issues/784

After a seemingly successful install, I reboot as requested. Reboot isn't successful, as it gets stuck with a blinking cursor (I think it's X crashing but I'm not sure, I'm new to Linux). After that, I don't know what to do. So, taking my inexperience into account, I just play it safe and load the latest system snapshot.

Here is my debug output:
```
Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Linuxmint
Release: vera
Kernel: 5.15.0-71-generic

---------------- DisplayLink info ----------------

Driver version: 
DisplayLink service status: 
EVDI service version: /sys/devices/evdi/version not found

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: GT2
[GeForce
VGA: Intel Corporation CoffeeLake-H GT2 [UHD Graphics 630]
NVIDIA Corporation GP107M [GeForce GTX 1050 Mobile] (rev a1)
VGA (3D): 
X11 version: 21.1.4-2ubuntu1.7~22.04.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x44 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 3 associated providers: 1 name:modesetting
Provider 1: id: 0x1f3 cap: 0x2, Sink Output crtcs: 4 outputs: 3 associated providers: 1 name:NVIDIA-G0
```

I get `DisplayLink service status: EVDI service version: /sys/devices/evdi/version not found`, just like the previously mention user.

Thanks in advance!

boris-ankov, 2023-04-26:
Same issue here.. this is frustrating.. (please view my comments in #826 & #785)

iamdey, 2023-05-30:
This is probably due to an error when building evdi, see #821 
I think this should be closed as duplicate

Barabazs, 2023-06-04:
@losocio it appears that there is a problem with installing evdi:
`EVDI service version: /sys/devices/evdi/version not found`

I would suggest retrying the install procedure and copy pasting the entire install terminal output here.

mzhk, 2023-06-07:
See https://github.com/AdnanHodzic/displaylink-debian/issues/828#issuecomment-1580946097

losocio, 2023-06-22:
I'll try to fix it later. Busy rn

closed 2023-06-22 by losocio