AdnanHodzic/displaylink-debian #826: EVDI service version: /sys/devices/evdi/version not found

jotapikie, 2023-04-26:
Hey! I already tried multiple ways to install the evdi kernel but none of them worked. I kindly ask for your support to help me with this problem. I attached the installation output and --debug output for easy reading. There is an error on line 86 of the installation output. I am also attaching the /var/lib/dkms/evdi/1.13.1/build/make.log. Any additional  information just ask. Thanks in advance.
[debug.txt](https://github.com/AdnanHodzic/displaylink-debian/files/11332505/debug.txt)
[installation_output.txt](https://github.com/AdnanHodzic/displaylink-debian/files/11332506/installation_output.txt)
[make.log](https://github.com/AdnanHodzic/displaylink-debian/files/11332507/make.log)

losocio, 2023-04-26:
Same issue on my part, I posted an issue right before you (issue number 825, yours is 826).

iamdey, 2023-05-30:
This is probably due to an error when building evdi, see #821 
I think this should be closed as duplicate

mzhk, 2023-06-07:
See https://github.com/AdnanHodzic/displaylink-debian/issues/828#issuecomment-1580946097

JeromeHen, 2023-07-16:
I got the same issue on kernel 6.3.0-kali1-amd64. I managed to narrow it down to a [missing i915 kernel module](https://www.reddit.com/r/debian/comments/11qvk0y/bookworm_possible_missing_firmware/).

I waited for a bit and now thanks to the addition of this module to the rolling distro, I was able to `sudo apt install evdi-dkms`, reinstall displaylink-debian and change /etc/X11/xorg.conf.d/20-displaylink.conf like mentioned [here](https://github.com/AdnanHodzic/displaylink-debian/issues/228#issuecomment-467889348).

Carefull tho, I had an instance where I got a **kernel panic not syncing vfs** after a bad update by installing evdi-dkms myself. I had to load in recovery to uninstall displaylink-driver, `sudo apt clean` and reboot.